softwarecrash/EPEver2MQTT #8: Uhrzeit falsch

mymeyer, 2023-04-23:
Hallo, bei mir wird die Uhrzeit falsch angezeigt !  (2013-01-02) Woran kann das liegen ?

MyMeyer

softwarecrash, 2023-04-23:
Hallo,
vermutlich weil die gerätezeit auf dieses datum (nicht uhrzeit) gestellt ist

mymeyer, 2023-04-23:
> Hallo, vermutlich weil die gerätezeit auf dieses datum (nicht uhrzeit) gestellt ist

Vom EPever ? Wie kann man denn das ändern ?
🤔

softwarecrash, 2023-04-23:
klick mal auf "set device from computer" dann wird die aktuelle pc zeit genommen und auf das gerät übertragen
[image]

mymeyer, 2023-04-23:
Merci Perfekt 👍

Emmpunkt, 2023-06-26:
Ich klink mich hier mal ein, obwohl geschlossen.
Wie mache ich das wenn ich mehrere Geräte habe?
Bei mir stellt er nur Uhrzeit und Datum auf der ersten Adresse.
Und hat die Zeit überhaupt eine Funktion?
Gruß M.
Danke fürs entwickeln und teilen, hatte vorher was auf NodeRed-USB Basis probiert, hat nicht funktioniert ;)

softwarecrash, 2023-07-02:
> Ich klink mich hier mal ein, obwohl geschlossen. Wie mache ich das wenn ich mehrere Geräte habe? Bei mir stellt er nur Uhrzeit und Datum auf der ersten Adresse. Und hat die Zeit überhaupt eine [...]

die zeit hat die funktion des tag/nacht schaltens und produktionsdaten richtig erfassen.
normal funktioniert das ganz gut mit der uhrzeit

robi2mow, 2023-07-15:
Bei mir lässt sich die Uhrzeit von Gerät ID2 auch nicht einstellen, sieht so aus, als würde immer ID1 gestellt.
Habe 2x 5415AN mit ID1 und ID2 in Verwendung.
Sonst funktioniert das Modul super, danke für die Arbeit!
Liebe Grüße 
Gerhard

softwarecrash, 2023-07-15:
@robi2mow @Emmpunkt 
probiert mal bitte diese, ich kann es aktuell leider nicht in echt testen
[EPEver2MQTT_1.0.3.1.zip](https://github.com/softwarecrash/EPEver2MQTT/files/12060912/EPEver2MQTT_1.0.3.1.zip)

Emmpunkt, 2023-07-15:
Ich hab mittlerweile 4 Regler.
Vers. 1.0.3.1 Es wurde die Uhrzeit von Regler 1 und 3 gestellt (1 war schon)
Vers. 1.0.3.2 Alle Regler haben die gleiche Uhrzeit. Ob er jetzt nur 2 und 4 gestellt hat, kann ich nicht sagen.;)

Das schalten der Load-Ausgänge funktionierte in allen Versionen. 
(Aktuell hängt aber nur an einem Regler auch wirklich was dran)

Was mir noch aufgefallen ist: Der Button Pfeil links funktioniert nicht immer, er funktioniert aber wenn man erst Pfeil rechts klickt und dann links. Browser: Chrome

softwarecrash, 2023-07-15:
Ob alle Uhrzeiten gesetzt werden kannst du testen indem du die PC Uhrzeit oder Handy verstellst, daher holt er sich die Zeit.

Mit den Pfeilen das kann ich gerade nicht ganz nachvollziehen, Standart fängt er beim ersten any nach rechts geht's aufwärts, von 1 nach links zu 4 wechseln geht nicht (keine Rotation)

Emmpunkt, 2023-07-15:
Uhrzeit funktioniert (1.0.3.2)
Pfeile:
Beispiel: Interface steht auf 4. Ich gehe ins Settings Menu und wieder Back. Klicke ich auf Links passiert nichts (Auch die Daten werden von 4 angezeigt, ist also nicht nur der Text) Klicke ich rechts, Springt er zu Regler 2.( Als ob gespeichert ist, das er vorher auf 1 stand)

softwarecrash, 2023-07-15:
Super, Zeit erledigt.
Mit den Pfeilen muss ich mir später Mal genauer ansehen was da nicht richtig läuft, hab das nie getestet mit ins Menü gehen und zurück.

robi2mow, 2023-07-15:
Danke, habs auch grad getestet. bei 1.0.3.1 hat es noch nicht funktioniert, bei 1.0.3.2 hat es sofort funktioniert.
Die Pfeile haben sich bei mir auch komisch verhalten, nach links gings oft nicht. Bei der neuen Version geht es jetzt immer, habe aber nur 2 Geräte.
Liebe Grüße 
Gerhard

softwarecrash, 2023-07-16:
fixed

closed 2023-07-16 by softwarecrash