softwarecrash/EPEver2MQTT #13: [Bug]: MQTT client show previous values on start

murfy76, 2023-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

When I start MQTT client, it will show incorrect values. After first update it will show correct values. For example SOC at MQTT client start is always 100 and after first update from ESP it will show actual value. Tested on "IoT MQTT panel" and "MQTT Explorer". Using JSON output, Mosquito broker. My others MQTT endpoints not behave like this. They don't display any value until they receive update. Thank you.

      

### Screenshots / Fotos

[image]

### Steps To Reproduce

1. start MQTT explorer
2. inscpect SOC value - 100
3. wait for update from ESP 

### Version

1.x.x and above

### Relevant livejson output

```json
{"LiveData":{"SOLAR_VOLTS":65.87,"SOLAR_AMPS":2.41,"SOLAR_WATTS":159.06,"BATT_VOLTS":26.6,"BATT_AMPS":5.98,"BATT_WATTS":159.06,"LOAD_VOLTS":26.6,"LOAD_AMPS":0,"LOAD_WATTS":0,"BATTERY_SOC":92},"StatsData":{"SOLAR_MAX":85.51,"SOLAR_MIN":0.04,"BATT_MAX":26.75,"BATT_MIN":25.59,"CONS_ENERGY_DAY":0,"CONS_ENGERY_MON":0,"CONS_ENGERY_YEAR":7.62,"CONS_ENGERY_TOT":16.56,"GEN_ENERGY_DAY":0.22,"GEN_ENERGY_MON":0.22,"GEN_ENERGY_YEAR":69.55,"GEN_ENERGY_TOT":100.8,"CO2_REDUCTION":0.1},"DEVICE_NAME":"epever","DEVICE_QUANTITY":1,"DEVICE_TIME":1685619505,"DEVICE_TEMPERATURE":24.13,"DEVICE_FREE_HEAP":37016,"DEVICE_JSON_MEMORY":628,"ESP_VCC":2.98,"BATTERY_TEMPERATURE":15.75,"LOAD_STATE":true,"BATT_VOLT_STATUS":"Normal","BATT_TEMP":"Normal","CHARGER_INPUT_STATUS":"Normal","CHARGER_MODE":"Boost"}
```

### What browsers are you seeing the problem on?

Firefox

murfy76, 2023-06-02:
Hello ManuA,

I'm not aware of any other device posted to this MQTT topic. I'm having only one Epever2Mqtt ESP. I can reproduce this even with other values, like SOLAR_AMPS. First value is always :0.13:

[image]

[image]

Thanks.

Peter

murfy76, 2023-06-02:
It's EPsolar XTRA 4415N and current EPEver2MQTT v1.0.1. Thanks.

softwarecrash, 2023-06-02:
when comes the wrong data? when you restart the esp?
its actual a bit hard to reproduce this behavior

murfy76, 2023-06-03:
Wrong data comes just first time when "MQTT explorer" or "IOT MQTT panel" connect to Mosquito broker. It looks to me like some persistent data, which are always the same when MQTT client first connect to broker, but I don't know where it comes from. If you cannot reproduce it, you can close this bug. Thank you anyway. 

[image]

softwarecrash, 2023-06-04:
so i have double checked the data and cant find anything who generates this wrong data, the only part that have a LWT or retain  is the alive.

closed 2023-06-04 by softwarecrash

murfy76, 2023-06-20:
Hello. I found cause of this strange behavior. I've deleted (with MQTT Explorer) all epever2mqtt topic and it's gone - no more previous values on startup. Thanks.

softwarecrash, 2023-06-20:
thats realy strange, but when its work now, great :)