softwarecrash/EPEver2MQTT #16: [Bug]: BATT_AMPS & BATT_WATTS wrong readings

MatchoLM, 2023-07-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

BATT_AMPS & BATT_WATTS in MQTT and also in web ui did not fit to real battery values, actual values are about total current & power production of Epever regulator. Without used LOAD the values are okay because current or power going only to battery, but in case LOAD is used like in my case the total current or power is devided to load and battery, so correct calculation for battery values should be total amps - load or total watts - load. In case solar panells are not producing enought power, current going from battery to load and on battery should be negative reading of current or power.  

### Screenshots / Fotos

[image]

### Version

1.x.x and above

### Relevant livejson output

```json
{"LiveData":{"SOLAR_VOLTS":95.22,"SOLAR_AMPS":0.95,"SOLAR_WATTS":90.95,"BATT_VOLTS":25.84,"BATT_AMPS":3.52,"BATT_WATTS":90.95,"LOAD_VOLTS":25.84,"LOAD_AMPS":3.47,"LOAD_WATTS":89.66,"BATTERY_SOC":55},"StatsData":{"SOLAR_MAX":109.85,"SOLAR_MIN":4.96,"BATT_MAX":26.2,"BATT_MIN":25.12,"CONS_ENERGY_DAY":0.83,"CONS_ENGERY_MON":7.46,"CONS_ENGERY_YEAR":7.46,"CONS_ENGERY_TOT":7.76,"GEN_ENERGY_DAY":0.26,"GEN_ENERGY_MON":7.1,"GEN_ENERGY_YEAR":7.1,"GEN_ENERGY_TOT":7.52,"CO2_REDUCTION":0},"DEVICE_NAME":"EPEver2MQTT","DEVICE_QUANTITY":1,"DEVICE_TIME":1690359108,"DEVICE_TEMPERATURE":32.07,"DEVICE_FREE_HEAP":35528,"DEVICE_JSON_MEMORY":653,"ESP_VCC":2.991,"Wifi_RSSI":-66,"BATTERY_TEMPERATURE":20.55,"LOAD_STATE":true,"BATT_VOLT_STATUS":"Normal","BATT_TEMP":"Normal","CHARGER_INPUT_STATUS":"Normal","CHARGER_MODE":"Boost"}
```

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2023-07-26:
we dont do any calculations on the ESP, all you see is what we get from the EPever device
check the coresponding datasheet:
https://github.com/softwarecrash/EPEver2MQTT/blob/main/Protocol/ControllerProtocolV2.3.pdf

MatchoLM, 2023-07-26:
thx for reply, i checked datasheet and i find there are two adresses for battery current 331B and 331C can we test it ? please see reading from Solar Station MonitorV1.95 got me calculatred value
 [image]

softwarecrash, 2023-07-26:
I dont know solar Station Monitor, ITS possibile that this Males calculations based in the Data
These two Data field dont need Testing, its marked with H and L that mean that this is a high and Low Byte and need shiftet to one dataset.
But I have seen some curios Data structure ordering, il Check it Later.

Unfourtanley i dont have a epever device Here, @all-solutions have one and can Check it.
We must wait for him

MatchoLM, 2023-07-26:
-1,12A that was situation when i did not get enought power from solar and current go from battery to load. now should be on battery -3.6A but shows 0A -battery shows now value of total amps generated by charger/regulator
[image]

softwarecrash, 2023-07-26:
now its clearer, let me check that the next days

MatchoLM, 2023-07-26:
yeah sure thx so much :) i found picture from regulator, just to be 100% clear , solar panel got just few watts or amps so load was 2,8A , from battery was taken -2,4A 
[image]

softwarecrash, 2023-07-26:
so understand right,
the batter A must shift to load
and battery A need a new value from 311B/C
right?

MatchoLM, 2023-07-26:
produced Amps are divided to battery and loads like example production is 20A loads is 5A then battery get 15A. if is production Amps lower than loads Amps like example 2A producced, loads 5A  then battery should show negative readings -3A because current from production is not enought and must be taken from battery

softwarecrash, 2023-07-26:
[EPEver2MQTT_d1_mini_1.1.1-1.zip](https://github.com/softwarecrash/EPEver2MQTT/files/12176985/EPEver2MQTT_d1_mini_1.1.1-1.zip)
check this please, batt amp have now the value from 311B/C

MatchoLM, 2023-07-26:
for now it looks better, i will see tomorrow when sun came out and start some production :) 
[image]

softwarecrash, 2023-07-26:
ok, watt cannot correct, that must be a calculated value or removed, i never seen this as negative

MatchoLM, 2023-07-26:
could be if you dont use load terminals on charger. benefit of using load terminals is you can monitor your real load what is not possible if you connect load directly to battery, and also when is load connected directly to battery you can not switch off load via regulator and on battery stats you will get Amps and Watts from battery + loads together. on my epever xtra4415n there are loads terminals up to 40A so no problem to use them for 500/1000w inverter :) 
[image]

softwarecrash, 2023-07-26:
[EPEver2MQTT_d1_mini_1.1.1-2.zip](https://github.com/softwarecrash/EPEver2MQTT/files/12177127/EPEver2MQTT_d1_mini_1.1.1-2.zip)
for the case the values are right, here the .1-2 with corrected amp and watt values for the battery
please report

MatchoLM, 2023-07-26:
in this case watts are calculated with wrong decimal point -95.49W should be ok. 
[image]

softwarecrash, 2023-07-26:
what the heck? too late for calculations, i hope this will fit
[EPEver2MQTT_d1_mini_1.1.1-3.zip](https://github.com/softwarecrash/EPEver2MQTT/files/12177196/EPEver2MQTT_d1_mini_1.1.1-3.zip)

MatchoLM, 2023-07-26:
yeah now is okay :) i let you know tomorrow how it works with produced power from solar panel, anyhow now is too late and both of us need some beer haha. Thank you for quick responses and findings.
[image]

softwarecrash, 2023-07-26:
jep,
just for test, will the watt with this correct and with 2 decimal places
[EPEver2MQTT_d1_mini_1.1.1-4.zip](https://github.com/softwarecrash/EPEver2MQTT/files/12177289/EPEver2MQTT_d1_mini_1.1.1-4.zip)

MatchoLM, 2023-07-26:
this version does wrong decimals
[image]

softwarecrash, 2023-07-26:
lol what a mess... my calculator does it right
last try for now
[EPEver2MQTT_d1_mini_1.1.1-5.zip](https://github.com/softwarecrash/EPEver2MQTT/files/12177380/EPEver2MQTT_d1_mini_1.1.1-5.zip)

MatchoLM, 2023-07-26:
it is too late also for calculator haha 
[image]

softwarecrash, 2023-07-27:
the decimal thing i will fix later....
so what is production for data? can you pull out the data points from the pdf and sort it to the places?

MatchoLM, 2023-07-27:
"production" data will show total Amps produced by regulator and i like to have it to see also total produced Apms, that would be same value like was before on battery position in version 1.1.0 like i call it "total" on first post picture

softwarecrash, 2023-07-27:
theoretical possibile, but its like the watt on battery, its not the best way to calculate some data.
so we need a new layout ore ordering / sorting of the avaible data. 
so when you have a idea, make a picture with a sorting / layout with places marked witht the hex data field and value like (311C)W

MatchoLM, 2023-07-28:
i checked with Controller Protocol V2.v3.pdf and also with Epever Solar Station Monitor 1.95 but i did not find corresponding value for it. so i see only possibility to use previous code of battery colum for new one "production" which should be easiest way or then is second option which is from my point of you more precise, from v1.1.1-5 take values of battery and load and calculate them together batt+load and that is equal to total production of regulator. I expect calculated value (W) should be different then readings W from solar because conversion efeciency of regulator is flexible.
[image]

softwarecrash, 2023-07-28:
thanks.
i must fix another thing, then i will look that i make the data right

MatchoLM, 2023-07-28:
yeah sure , thank you ! :)

softwarecrash, 2023-07-31:
i have revertet something, and corrected the battery Ampere value.
can you check the battery Watt value when the batery ampere goes negativ?

in wrong case i think to remove the battery watt data, these are similar to the solar watt data and the epever doesnt provide the battery watt value
[EPEver2MQTT_esp01_1m_1.1.1-6.zip](https://github.com/softwarecrash/EPEver2MQTT/files/12213625/EPEver2MQTT_esp01_1m_1.1.1-6.zip)

MatchoLM, 2023-07-31:
can you please upload d1_mini version ?

softwarecrash, 2023-07-31:
ops, my fault. here with mini
[EPEver2MQTT_d1_mini_1.1.1-7.zip](https://github.com/softwarecrash/EPEver2MQTT/files/12214078/EPEver2MQTT_d1_mini_1.1.1-7.zip)

MatchoLM, 2023-07-31:
i checked it and watts are not correct on both scenarios in positive readings there value from solar and in negative scenario is issue same, value from solar. Correct watts was in version 1.1.1-3 there was just small thing about to many numbers after dot what is from my point of you no any mistake. btw why do you like to avoid some calculations in your code?
[image]
[image]

softwarecrash, 2023-08-02:
im on it.
why avoid calculations? its a relativ big code impact, or other the cpu need a lot of cycles to calculate two float values into one and correct the decimal places, but i look that i can make it.
on the other side, i get now a epever and rewrite some things

MatchoLM, 2023-08-03:
that sounds great :) if you will need any help with testing or something just let me know.

softwarecrash, 2023-08-04:
i think i have managed the watt value and the negative battery.
please check
[EPEver2MQTT_d1_mini_1.1.1-8.zip](https://github.com/softwarecrash/EPEver2MQTT/files/12262226/EPEver2MQTT_d1_mini_1.1.1-8.zip)

MatchoLM, 2023-08-04:
it looks okey :) good job :) 

[image]

softwarecrash, 2023-08-04:
great, it will comes offizial with the next release

MatchoLM, 2023-08-04:
thx so much :)

closed 2023-08-04 by MatchoLM