softwarecrash/EPEver2MQTT #21: [Bug]: Typo: Temperature shown a Gemperature Vers. 2.0.0

Emmpunkt, 2023-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Title says all, nothing more to tell...

### Screenshots / Fotos

[image]

### Version

1.x.x and above

### Relevant livejson output

```json
Don.t know what i should paste here
```

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2023-08-21:
thank you for the hint.
it will be fixed with the next release

softwarecrash, 2023-09-01:
Fixed with V2.0.1

closed 2023-09-01 by softwarecrash