softwarecrash/EPEver2MQTT #22: Reading HTML from PROGMEM in htmlProzessor force exception.

rolfeisenhut, 2023-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

When compiling the Original Sourcecode and uploading it into the device, i always get un exception within htmlProzessor.h code, reporting there is a problem accessing the PROGMEM Code from the html page, seems to be HEAD and/or Footer.
After using FPSTR(...) method it worked.
htmlprozessor.h
if (var == F("pre_head_template"))
        return (FPSTR(HTML_HEAD));
    if (var == F("pre_foot_template"))    
        return (FPSTR(HTML_FOOT));
    if (var == F("pre_software_version"))
Have a look into https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html
RAM usage remains the same, but exception disappears. 

Please adjust your source code accordingly. 

regards Rolf 

  

### Screenshots / Fotos

empty screen ...

### Steps To Reproduce

Calling Main web-page and see serial monitor in visuao code pio  project .
Hope code can help for more details.
Latest version 2.0.0. 
Got same issue in some previous version, causing the issue with thew footer html.

Regards Rolf

``
16:25:22.157 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------
16:25:22.163 > 
16:25:22.163 > Exception (3):
16:25:22.163 > epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4025c531 depc=0x00000000
16:25:22.239 > 
16:25:22.239 > LoadStoreError: Processor internal physical address or data error during load or store
16:25:22.239 >   excvaddr=0x4025c531 in sleep_reset_analog_rtcreg_8266 at ??:?
16:25:22.239 > 
16:25:22.239 > >>>stack>>>
16:25:22.239 > 
16:25:22.239 > ctx: sys
16:25:22.634 > sp: 3fffe9d0 end: 3fffffb0 offset: 0190
16:25:22.634 > 3fffeb60:  3fffec00 3fffeb90 00000000 4021c5b0  
16:25:22.634 > 3fffeb70:  4025c531 3fffec00 3fffeb90 4021bdd7  
16:25:22.634 > 3fffeb80:  3fffebe1 3fffec00 3fffebf4 40201977  
16:25:22.634 > 3fffeb90:  3fff4500 0011001f 00ffebf4 4021bf88  
16:25:22.634 > 3fffeba0:  3fffebe1 3fff48f2 3fffebf4 000003b8  
16:25:22.634 > 3fffebb0:  3fffebe1 3fff48f2 3fffebf4 40220250  
16:25:22.634 > 3fffebc0:  3fff47c0 3fff487c 3fff477c 40211180  
16:25:22.634 > 3fffebd0:  5f657270 64616568 6d65745f 74616c70  
16:25:22.634 > 3fffebe0:  3fff0065 4021005d 00000020 40100d3e  
16:25:22.634 > 3fffebf0:  3fffec60 3fff3600 0011001f 0021bf88  
16:25:22.634 > 3fffec00:  3fff366c 0011001f 8021bf88 40100a83  
16:25:22.634 > 3fffec10:  3fff4904 3fff48f2 3fff4ca9 3fff48f2  
16:25:22.634 > 3fffec20:  000003b8 000003b7 3fff4ca9 3fff477c  
16:25:22.634 > 3fffec30:  3fff48f3 00000000 00000020 40100d3e  
16:25:24.601 > 3fffec40:  3fffec6c 00000005 3fffec6c 3fff477c  
16:25:24.601 > 3fffec50:  3fff47c0 3fff487c 00000070 40211429  
16:25:24.601 > 3fffec60:  6e756800 0064656b 00ff4784 3fff4700  
[... 410 lines cut ...]
16:25:27.025 > 0x40215ab8 in esp8266::MDNSImplementation::MDNSResponder::_announce(bool, bool) at ??:?
16:25:27.025 > 0x4021a36c in ModbusMaster::readHoldingRegisters(unsigned short, unsigned short) at ??:?
16:25:27.025 > 0x40202a84 in getEpData(int) at ??:?
16:25:27.025 > 0x402205bc in PubSubClient::connected() at ??:?
16:25:27.025 > 0x402097f2 in PubSubClient::loop() at ??:?
16:25:27.025 > 0x40208e34 in epWorker() at ??:?
16:25:27.025 > 0x40208f14 in loop at ??:?
16:25:27.025 > 0x4021d27c in loop_wrapper() at core_esp8266_main.cpp:?
16:25:27.025 >
16:25:27.025 >
16:25:27.025 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------

### Version

1.x.x and above

### Relevant livejson output

```json
...
```

### What browsers are you seeing the problem on?

Chrome

rolfeisenhut, 2023-08-21:
I found another issue.
In case of no EP is connected, the new epwroker loops in a very hight frequency, 5 - 10 time a second, which is useless.
Better put a delay of 1-5 sec  after notify client in case of getepdata()  returns false.

Rolf

softwarecrash, 2023-08-21:
Hello,
thanks for this great report.
thats interesting, i never had problems with the progmem.
so i have made the changes, please check the branch "bugfix-2.0.0" for the progmem error and the notification speed when no device is connected

softwarecrash, 2023-09-01:
Fixed with V2.0.1

closed 2023-09-01 by softwarecrash