softwarecrash/EPEver2MQTT #25: [Bug]: Uhrzeit verstellt sich sporadisch

Jogi2020, 2023-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Seit ich die Version 2.0.x installiert habe, habe ich Probleme mit der Urzeit.
Am 07.09. wurde z.B. um 11Uhr die Zeit am EPEver au 0Uhr gesetzt. Das hat zur Folge das GEN_DAY auf 0 zurück gesetzt wird, wodurch die täglichen Werte nicht mehr stimmen. Meist geht die Uhr des EPEver nur etwas vor, so dass der EPEver den Wert für GEN_DAY schon um 23:59Uhr auf 0 setzt. Das hat zur folge, dass der letzte Wert, der vor 0Uhr per MQTT übertragen wir 0 ist und dadurch meine grafische Anzeige für den letzten Tag ebenfalls 0 anzeigt und nicht die Tagesproduktion.
Wenn ich die Uhrzeit dann über die Weboberfläche des EPEver2MQTT auf dem EPEver neu stellen will muss ich meist mehrfach auf "set device time von Computer" drücken bis die Zeit vom EPEver wieder stimmt.

### Screenshots / Fotos

[image]

### Version

1.x.x and above

### Relevant livejson output

```json
{
    "DEVICE_NAME": "EPEver2MQTT",
    "EP_1": {
        "LiveData": {
            "CONNECTION": 0,
            "DEVICE_NUM": "1",
            "DEVICE_TIME": 1694127885,
            "DEVICE_TEMP": 26.72,
            "SOLAR_V": 0.17,
            "SOLAR_A": 0,
            "SOLAR_W": 0,
            "BATT_SOC": 57,
            "BATT_V": 25.31,
            "BATT_A": 0,
            "BATT_W": 0,
            "BATT_STATE": "Normal",
            "BATT_TEMP": 24.58,
            "BATT_TEMP_STATE": "Normal",
            "LOAD_V": 25.31,
            "LOAD_A": 0,
            "LOAD_W": 0,
            "LOAD_STATE": true,
            "CHARGER_STATE": "Normal",
            "CHARGER_MODE": "Off"
        },
        "StatsData": {
            "SOLAR_MAX": 0.18,
            "SOLAR_MIN": 0.04,
            "BATT_MAX": 25.61,
            "BATT_MIN": 25.21,
            "CONS_DAY": 0,
            "CONS_MON": 0.02,
            "CONS_YEAR": 3.41,
            "CONS_TOT": 3.41,
            "GEN_DAY": 0,
            "GEN_MON": 17.03,
            "GEN_YEAR": 484.62,
            "GEN_TOT": 484.62,
            "CO2_REDUCTION": 0.48
        },
        "DeviceData": {
            "BATTERY_TYPE": 5,
            "BATTERY_CAPACITY": 1.05,
            "TEMPERATURE_COMPENSATION": 3,
            "HIGH_VOLT_DISCONNECT": 29.6,
            "CHARGING_LIMIT_VOLTS": 29.2,
            "OVER_VOLTS_RECONNECT": 29.2,
            "EQUALIZATION_VOLTS": 29,
            "BOOST_VOLTS": 29,
            "FLOAT_VOLTS": 27.6,
            "BOOST_RECONNECT_VOLTS": 26.4,
            "LOW_VOLTS_RECONNECT": 25.6,
            "UNDER_VOLTS_RECOVER": 24.4,
            "UNDER_VOLTS_WARNING": 24,
            "LOW_VOLTS_DISCONNECT": 22.2,
            "DISCHARGING_LIMIT_VOLTS": 22
        }
    },
    "DEVICE_QUANTITY": 1,
    "DEVICE_FREE_HEAP": 30272,
    "DEVICE_FREE_JSON": 7186,
    "ESP_VCC": 3.369,
    "Wifi_RSSI": -64,
    "sw_version": "2.0.1"
}
```

### What browsers are you seeing the problem on?

Microsoft Edge

softwarecrash, 2023-09-08:
Hallo,
das du mehrfach auf den button für uhrzeit stellen drücken musst ist bekannt, das ist kein echter bug, das ist eher dem ganzen system geschuldet, aber ich schau mal ob ich das evt elleganter hinbekomme, kann aber nichts versprechen.

das sich die uhrzeit von selbst verstellt ist eigentlich fast unmöglich, da einzig durch den button dieses setzen der uhrzeit ausgelöst wird, aber auch da schaue ich nochmal nach.

das die uhrzeit vor oder nachgeht darauf habe ich keinen einfluss

softwarecrash, 2023-09-10:
ich hab das jetzt eine weile im test laufen lassen, leider kann ich das verhalten nicht nachvolllziehen oder einen fehler  finden.

Jogi2020, 2023-09-10:
Vielen Dank für deine Mühe, seit meinem Post läuft es im Moment bei mir auch. Vielleicht hat ja auch mein EPEver ein Problem und verstellt durch irgendwas die Zeit. Das ging bei mir seit Januar einwandfrei, nach dem Update auf 2.0 hatte ich dann festgestellt, dass meine grafische Auswertung keinen Tagesertrag mehr anzeigt und auf Null steht. Ich hatte dann festgestellt, das im EPEver die Zeit nicht stimmt. Das kann natürlich auch am EPEver selbst liegen.
Wenn ich dich richtig verstanden habe wird die Zeit nur übertragen, wenn der Button auf der Weboberfläche gedrückt wird.
Wäre es nicht eine Idee, das der ESP die Uhrzeit von einem Zeitserver holt und dann täglich per Programm überträgt?

softwarecrash, 2023-09-10:
evt kann es passiert sein bei dem flashen oder sonstwas das der ep irgendwas aufgeschnappt hat was er für eine zeitangabe gehalten hat und übernommen.
genau die zeit wird einzig durch den button gesetzt, die idee mit NTP gab es schon öfter, aber einerseits ist die uhr de ep relativ genau, andererseits müsste man dann eine config bauen damit jeder seine zeitzone sommer/winter usw angeben kann. den ganzen aufwand nur um den ep auf ein paar sekunden im monat zu korrigieren finde ich übertrieben

closed 2023-10-25 by github-actions[bot]