softwarecrash/EPEver2MQTT #28: [Bug]: Setzen der Zeit

Jogi2020, 2023-11-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Seit der Version 2.1.0 funktioniert bei mir das setzen der Zeit nicht mehr richtig.
Ich lebe in der Zeitzone UTC+1 Berlin. Wenn ich z.B. um 22:Uhr bei mir die Zeit übertrage, wird anschließend 21: Uhr angezeigt.
Das ist bei mir mit dieser Zeitzone reproduzierbar. Wechsel ich auf eine andere Zeitzone z.B. UTC+2 wird die Zeit korrekt übertragen.
Übrigen seit der Version 2.1.0 läuft die Urzeit bei mir wieder stabil und wird nicht mehr Tagsüber ca. 11Uhr auf 0 gesetzt.

### Screenshots / Fotos

[image]
[image]

### Steps To Reproduce

Der PC zeigt 22 Uhr
setze die Zeit mit UTC+1: es wir die UTC angezeigt 21 Uhr
setze die Zeit auf UTC+2 und setze die zeit auf dem ESP: die Zeit wird korrekt angezeigt (UTC+2 gleich 23 Uhr
setze die Zeit am PC auf automatisch Die Zeit vom D1 wird richtig angezeigt 22 Uhr
Setze die Zeit des D1 es wird 21 Uhr angezeigt

### Version

1.x.x and above

### Relevant livejson output

```json
Kein
```

### What browsers are you seeing the problem on?

Chrome, Microsoft Edge

softwarecrash, 2023-11-28:
Danke,
ich schau mal das ich das nochmal korrigiere

softwarecrash, 2023-11-28:
@Jogi2020 
probiere bitte diese version
[EPEver2MQTT_d1_mini_2.1.1A-2.zip](https://github.com/softwarecrash/EPEver2MQTT/files/13493315/EPEver2MQTT_d1_mini_2.1.1A-2.zip)

closed 2023-12-05 by softwarecrash