softwarecrash/EPEver2MQTT #29: [FEATURE] HA Discovery GEN_TOT für Statistik

subivoodoo, 2023-12-10:
### Is your feature request related to a problem? Please describe

Die Home Assistant Integration mittels MQTT Discovery funktioniert gut und zeigt alle Werte wie gewünscht.

Jedoch kann z.B. der Sensor sensor.epever2mqtt_ep_1_gen_tot in HA nicht als Solarproduktion oder Batteriespeicher ausgewählt werden im Energy Dashboard und es werden auch keine Statistiken gespeichert.

### Describe the solution you'd like

Damit die generierte Energie im HA Energy Dashboard einfach genutzt/ausgewählt werden kann, wäre es gut wenn via Discovery die `state_class: energy` resp. `device_class: total_increasing` mitgegeben werden würde beim Total Energie Sensor.

### Describe alternatives you've considered

Als Workaround kann ein zusätzliches Template angelegt werden z.B. wie folgt:

```
template:
  - sensor:
    - name: "Solarenergie in Batterie"
      unique_id: epever.charge_total_sensor
      state: >-
          {{ 
            [ states('sensor.epever2mqtt_ep_1_gen_tot'), 
            ] | map('float') | sum
          }}
      unit_of_measurement: kWh
      device_class: energy
      state_class: total_increasing
      attributes:
        last_reset: "1970-01-01T00:00:00+00:00"
      availability: >-
          {{ 
            [ states('sensor.epever2mqtt_ep_1_gen_tot'), 
            ] | map('is_number') | min
          }}
```

softwarecrash, 2023-12-11:
Hallo,
aktuell wird bei GEN_TOT die device_class mit energy angegeben, das könnte ich zu total_incerasing ändern.
ob das funktioniert kannst du gern mal testen wenn du mit dem mqttexplorer auf deinen mqtt broker schaust dort unter homeassistant/sensors/EP... den GEN_TOT punkt suchst und in den json daten das entsprechend änderst

softwarecrash, 2023-12-11:
danke für die tests.
stat_class ist aktuell garnicht implementiert, da ich HA wenig bis garnicht kenne, bräuchte ich für alle betreffenden datenpunkte die richtigen stat zuweisungen, dann könnte ich mal schauen ob ich das noch unter bekomme

subivoodoo, 2023-12-11:
Ich bin auch kein HA Profi... Statistiken sind mehr oder weniger länger abgelegte Werte z.B. alle 5 Minuten quasi "für immer" abgespeichert damit darauf basierend u.a. das Energy Dashboard dargestellt werden kann und auch die Vorjahreswerte etc. historisiert sind. Sonst sind Werte in HA defaultmässig nur wenige Tage (7 oder 10) gespeichert und fliegen danach raus. Die vorhandenen Sensoren, zu welchen Statistiken gespeichert sind, sind in HA unter Entwicklertools>Statistik aufgelistet.

Es müssten sicher nicht alle Sensoren historisiert werden, die kWh wären jedoch schon sinnvoll.

Bei der AhoyDTU z.B. sind alle Energie-Sensoren (Wh und kWh) definiert als `state_class: total_increasing`, alle Spannungen/Ströme/Leistungen/Temperatur als `state_class: measurement` und Texte z.B. haben nix gesetzt... none könnte da gehen?

Jarnsen, 2023-12-11:
ne, none funktioniert nicht... das haben wir schon versucht

Jarnsen, 2023-12-12:
Rest ist mit SoftwareCrash besprochen, er "freut" sich das noch einzubauen ;-)

subivoodoo, 2023-12-12:
Cool danke! Es eilt aber überhaupt nicht, der Workaround funktioniert ja auch.

thitcher, 2023-12-13:
Kann man auch ganz einfach mit einem Riemann-Summenintegral zusammenbauen, ohne die JSON zu editieren.
Falls jemand (wie ich  :)  ) mit JSON und MQTT auf Kriegsfuß steht

softwarecrash, 2023-12-13:
@subivoodoo probier die mal, das sollte die entsprechenden einträge anlegen
[EPEver2MQTT_d1_mini_2.2.0A-1.zip](https://github.com/softwarecrash/EPEver2MQTT/files/13662593/EPEver2MQTT_d1_mini_2.2.0A-1.zip)

Kloana001, 2023-12-13:
[image]

Hallo, ich habe es gerade getestet und nun werden die Sensoren angezeigt...

Kloana001, 2023-12-13:
[image]
[image]

subivoodoo, 2023-12-13:
Kann ich auch bestätigen, jetzt ist alles ohne Workaround auswählbar, danke!!

[image]

softwarecrash, 2023-12-14:
fix with V2.2.1

closed 2023-12-14 by softwarecrash