softwarecrash/EPEver2MQTT #31: [Bug]: Ha zeigt Einheitenfehler einiger Entitäten

thitcher, 2023-12-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Ha meldet das für einige Entitäten die falsche Einheit verwendet wird:

Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:691
Integration: Sensor (documentation, issues)
First occurred: 03:00:45 (4 occurrences)
Last logged: 03:00:45

Entity sensor.epever_suden_co2_reduction (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 't' which is not a valid unit for the device class ('weight') it is using; expected one of ['µg', 'st', 'kg', 'lb', 'oz', 'mg', 'g']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.epever_suden_battery_capacity (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'Ah' which is not a valid unit for the device class ('energy_storage') it is using; expected one of ['Wh', 'GJ', 'MWh', 'MJ', 'kWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.epever_westen_co2_reduction (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 't' which is not a valid unit for the device class ('weight') it is using; expected one of ['µg', 'st', 'kg', 'lb', 'oz', 'mg', 'g']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.epever_westen_battery_capacity (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'Ah' which is not a valid unit for the device class ('energy_storage') it is using; expected one of ['Wh', 'GJ', 'MWh', 'MJ', 'kWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

### Screenshots / Fotos

[image]

### Steps To Reproduce

erscheint nach Neustart von HA in den logs

### Version

1.x.x and above

### Relevant livejson output

```json
n.A.
```

### What browsers are you seeing the problem on?

Firefox

Jarnsen, 2023-12-12:
das liegt daran das Home Assistant keine t und keine Ah normalerweise unterstützt. Die Alternative ist wir würden diese Entitäten nicht mit übertragen oder ohne Einheit bzw Unit. Dies sollte aber keinen Einfluss auf die Funktionalität haben.

softwarecrash, 2023-12-12:
HA kennt diese Einheiten einfach nicht. Da kann ich auch nichts machen.

closed 2023-12-12 by thitcher