softwarecrash/EPEver2MQTT #33: MQTT String Fehler

realmeto, 2024-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Hallo,
erst mal vielen Dank für die Entwicklung. Bei mir läuft das soweit super und stabiel derzeit mit der 2.2.0 mit 3 Devices.  Die Daten kommen alle über MQTT vollständig und fehlerfrei an.
In unregelmäßigen Abständen, zufällig und nicht nachproduzierbar kommt jedoch Datenmüll über mqtt. Scheinbar entsteht hier quatsch bei den Strings für die MQTT Übertragung. Ich Nutze IOBroker und muss die falschen Datenpunkte und Ordner regelmäßig löschen oder MQTT so einstellen, dass keine neuen Datenpunkte angelegt werden können. 

Ich hab mal ein Screenshot davon gemacht. Lässt sich das beheben? Ich hab noch keine Idee woran das liegen könnte.

### Screenshots / Fotos

[image]

### Version

1.x.x and above

### Relevant livejson output

```json
{"DEVICE_NAME":"EPEver2MQTT","EP_2":{"LiveData":{"CONNECTION":0,"DEVICE_NUM":"2","DEVICE_TIME":1704191470,"DEVICE_TEMP":8.77,"SOLAR_V":27.59,"SOLAR_A":0.25,"SOLAR_W":7.15,"BATT_SOC":55,"BATT_V":25.57,"BATT_A":0.28,"BATT_W":5.1,"BATT_STATE":"Normal","BATT_TEMP":25,"BATT_TEMP_STATE":"Normal","LOAD_V":0,"LOAD_A":0,"LOAD_W":0,"LOAD_STATE":false,"CHARGER_STATE":"Normal","CHARGER_MODE":"Boost"},"StatsData":{"SOLAR_MAX":74.38,"SOLAR_MIN":0.15,"BATT_MAX":25.58,"BATT_MIN":24.91,"CONS_DAY":0,"CONS_MON":0,"CONS_YEAR":0,"CONS_TOT":7.93,"GEN_DAY":0.01,"GEN_MON":0.07,"GEN_YEAR":0.07,"GEN_TOT":546.37,"CO2_REDUCTION":0.54},"DeviceData":{"BATTERY_TYPE":"User","BATTERY_CAPACITY":230,"TEMPERATURE_COMPENSATION":0,"HIGH_VOLT_DISCONNECT":29.2,"CHARGING_LIMIT_VOLTS":29,"OVER_VOLTS_RECONNECT":28.8,"EQUALIZATION_VOLTS":28.8,"BOOST_VOLTS":28.8,"FLOAT_VOLTS":28.8,"BOOST_RECONNECT_VOLTS":26.6,"LOW_VOLTS_RECONNECT":24,"UNDER_VOLTS_RECOVER":23.5,"UNDER_VOLTS_WARNING":23,"LOW_VOLTS_DISCONNECT":21.5,"DISCHARGING_LIMIT_VOLTS":20.5}},"DEVICE_QUANTITY":3,"DEVICE_FREE_HEAP":26536,"DEVICE_FREE_JSON":5551,"ESP_VCC":3.255,"Wifi_RSSI":-95,"sw_version":"2.2.0","EP_3":{"LiveData":{"CONNECTION":0,"DEVICE_NUM":"3","DEVICE_TIME":1704191473,"DEVICE_TEMP":8.57,"SOLAR_V":27.59,"SOLAR_A":0.6,"SOLAR_W":16.59,"BATT_SOC":55,"BATT_V":25.53,"BATT_A":0.66,"BATT_W":15.3,"BATT_STATE":"Normal","BATT_TEMP":6.22,"BATT_TEMP_STATE":"Normal","LOAD_V":0,"LOAD_A":0,"LOAD_W":0,"LOAD_STATE":false,"CHARGER_STATE":"Normal","CHARGER_MODE":"Boost"},"StatsData":{"SOLAR_MAX":38.08,"SOLAR_MIN":0.09,"BATT_MAX":25.54,"BATT_MIN":24.81,"CONS_DAY":0,"CONS_MON":0,"CONS_YEAR":0,"CONS_TOT":0,"GEN_DAY":0.03,"GEN_MON":0.17,"GEN_YEAR":0.17,"GEN_TOT":531.59,"CO2_REDUCTION":0.53},"DeviceData":{"BATTERY_TYPE":"User","BATTERY_CAPACITY":230,"TEMPERATURE_COMPENSATION":0,"HIGH_VOLT_DISCONNECT":29.2,"CHARGING_LIMIT_VOLTS":29,"OVER_VOLTS_RECONNECT":28.8,"EQUALIZATION_VOLTS":28.8,"BOOST_VOLTS":28.8,"FLOAT_VOLTS":28.8,"BOOST_RECONNECT_VOLTS":26.6,"LOW_VOLTS_RECONNECT":24,"UNDER_VOLTS_RECOVER":23.5,"UNDER_VOLTS_WARNING":23,"LOW_VOLTS_DISCONNECT":21.5,"DISCHARGING_LIMIT_VOLTS":20.5}},"EP_1":{"LiveData":{"CONNECTION":0,"DEVICE_NUM":"1","DEVICE_TIME":1704191489,"DEVICE_TEMP":8.88,"SOLAR_V":27.54,"SOLAR_A":0.32,"SOLAR_W":8.92,"BATT_SOC":54,"BATT_V":25.51,"BATT_A":0.36,"BATT_W":7.65,"BATT_STATE":"Normal","BATT_TEMP":6.17,"BATT_TEMP_STATE":"Normal","LOAD_V":0,"LOAD_A":0,"LOAD_W":0,"LOAD_STATE":false,"CHARGER_STATE":"Normal","CHARGER_MODE":"Boost"},"StatsData":{"SOLAR_MAX":38.57,"SOLAR_MIN":0.08,"BATT_MAX":25.53,"BATT_MIN":24.77,"CONS_DAY":0,"CONS_MON":0,"CONS_YEAR":0,"CONS_TOT":0,"GEN_DAY":0.01,"GEN_MON":0.1,"GEN_YEAR":0.1,"GEN_TOT":27.75,"CO2_REDUCTION":0.02},"DeviceData":{"BATTERY_TYPE":"User","BATTERY_CAPACITY":230,"TEMPERATURE_COMPENSATION":0,"HIGH_VOLT_DISCONNECT":29.2,"CHARGING_LIMIT_VOLTS":29,"OVER_VOLTS_RECONNECT":28.8,"EQUALIZATION_VOLTS":28.8,"BOOST_VOLTS":28.8,"FLOAT_VOLTS":28.8,"BOOST_RECONNECT_VOLTS":26.6,"LOW_VOLTS_RECONNECT":24,"UNDER_VOLTS_RECOVER":23.5,"UNDER_VOLTS_WARNING":23,"LOW_VOLTS_DISCONNECT":21.5,"DISCHARGING_LIMIT_VOLTS":20.5}}}
```

### What browsers are you seeing the problem on?

Firefox

softwarecrash, 2024-01-02:
Hallo,
das lässt sich aktuell nicht beheben, das passiert wenn wärend der übertragung kommunikationssörungen entstehen. oft tritt sowas auf wenn der wlan empfang schlecht ist oder gestört wird

realmeto, 2024-01-02:
OK. Danke für die schnelle Antwort!

closed 2024-01-02 by realmeto