softwarecrash/EPEver2MQTT #44: [Bug]: Extraneous character at the end of MQTT JSON message

sdalu, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Sometimes,  some extra characters are present at the end of the JSON MQTT message.

### Screenshots / Fotos

~~~
{"DEVICE_NAME":"EPEver2MQTT","EP_1":{"LiveData":{"CONNECTION":0,"DEVICE_NUM":"1","DEVICE_TIME":1715904852,"DEVICE_TEMP":13.93,"SOLAR_V":0.09,"SOLAR_A":0,"SOLAR_W":0,"BATT_SOC":63,"BATT_V":12.81,"BATT_A":0,"BATT_W":0,"BATT_STATE":"Normal","BATT_TEMP":14.07,"BATT_TEMP_STATE":"Normal","LOAD_V":0,"LOAD_A":0,"LOAD_W":0,"LOAD_STATE":false,"CHARGER_STATE":"Normal","CHARGER_MODE":"Off"},"StatsData":{"SOLAR_MAX":0.13,"SOLAR_MIN":0.09,"BATT_MAX":12.82,"BATT_MIN":12.73,"CONS_DAY":0,"CONS_MON":0,"CONS_YEAR":0,"CONS_TOT":4.08,"GEN_DAY":0,"GEN_MON":0,"GEN_YEAR":0,"GEN_TOT":3.18,"CO2_REDUCTION":0},"DeviceData":{"BATTERY_TYPE":"User","BATTERY_CAPACITY":200,"TEMPERATURE_COMPENSATION":3,"HIGH_VOLT_DISCONNECT":14.8,"CHARGING_LIMIT_VOLTS":14.6,"OVER_VOLTS_RECONNECT":14.6,"EQUALIZATION_VOLTS":14.5,"BOOST_VOLTS":14.5,"FLOAT_VOLTS":13.8,"BOOST_RECONNECT_VOLTS":13.2,"LOW_VOLTS_RECONNECT":12.8,"UNDER_VOLTS_RECOVER":12.2,"UNDER_VOLTS_WARNING":12,"LOW_VOLTS_DISCONNECT":11.1,"DISCHARGING_LIMIT_VOLTS":11}},"DEVI_QUANTITY":1,"DEVICE_FREE_HEAP":26728,"DEVICE_FREE_JSON":7229,"ESP_VCC":3.297,"Wifi_RSSI":-77,"sw_version":"2.3.0"}�
~~~

### Version

1.x.x and above

### Relevant livejson output

```json
.
```

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

softwarecrash, 2024-05-17:
hello,
never seen bevore this special char, sure your wifi and network are good and stable?

sdalu, 2024-05-17:
Yes, the network is good, and even if it wasn't, it shouldn't be an issue as MQTT is using TCP, which is reliable.
I saw it five or six times yesterday evening (I just installed it yesterday). I'll keep an eye on it and see if a pattern emerges.

sdalu, 2024-05-17:
Just got a new  incorrect MQTT JSON for the topic `DATA`, it was between preceded by a topic `Alive`=`true` and followed by topic `Alive` = `false`

softwarecrash, 2024-05-17:
Please try this firmware, it have improved speedup for json->mqtt and added a value for esp runtime, check this runtime value when wrong data comes up, it looks like your esp crashes. what power source you use?
[EPEver2MQTT_d1_mini_2.3.0A1.zip](https://github.com/softwarecrash/EPEver2MQTT/files/15348371/EPEver2MQTT_d1_mini_2.3.0A1.zip)

sdalu, 2024-05-17:
I'm using the RJ45 port of the MPPT Tracer AN Series

softwarecrash, 2024-05-17:
> I'm using the RJ45 port of the MPPT Tracer AN Series. (Firmware flashed)

some (the most) Tracer devices doesnt deliver enugh power for a ESP

softwarecrash, 2024-05-17:
btw your wifi is not gut with -77 dBm
[image]

sdalu, 2024-05-20:
So far, the new firmware seems to have resolved the problem.

sdalu, 2024-05-27:
Seems good :+1:

closed 2024-07-11 by github-actions[bot]