softwarecrash/EPEver2MQTT #46: [Bug]: Settings dialog incomplete

jurijzahn8019, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

- go to "/settingsedit"
- settings edit dialog is loaded incomlete

### Screenshots / Fotos

[image]

### Steps To Reproduce

- open webgui
- click settings
- click configure

### Version

1.x.x and above

### Relevant livejson output

```json
{
    "DEVICE_NAME": "EPEver2MQTT",
    "EP_1": {
        "LiveData": {
            "CONNECTION": 0,
            "DEVICE_NUM": "1",
            "DEVICE_TIME": 1718232492,
            "DEVICE_TEMP": 27.42,
            "SOLAR_V": 4.27,
            "SOLAR_A": 0,
            "SOLAR_W": 0,
            "BATT_SOC": 27,
            "BATT_V": 49.12,
            "BATT_A": 0,
            "BATT_W": 0,
            "BATT_STATE": "Normal",
            "BATT_TEMP": 24.78,
            "BATT_TEMP_STATE": "Normal",
            "LOAD_V": 0,
            "LOAD_A": 0,
            "LOAD_W": 0,
            "LOAD_STATE": false,
            "CHARGER_STATE": "Normal",
            "CHARGER_MODE": "Off"
        },
        "StatsData": {
            "SOLAR_MAX": 10.46,
            "SOLAR_MIN": 2.24,
            "BATT_MAX": 49.16,
            "BATT_MIN": 49.11,
            "CONS_DAY": 0,
            "CONS_MON": 0,
            "CONS_YEAR": 0,
            "CONS_TOT": 0,
            "GEN_DAY": 0,
            "GEN_MON": 0,
            "GEN_YEAR": 256.27,
            "GEN_TOT": 256.27,
            "CO2_REDUCTION": 0.25
        },
        "DeviceData": {
            "BATTERY_TYPE": "User",
            "BATTERY_CAPACITY": 71,
            "TEMPERATURE_COMPENSATION": 3,
            "HIGH_VOLT_DISCONNECT": 53,
            "CHARGING_LIMIT_VOLTS": 52.4,
            "OVER_VOLTS_RECONNECT": 51,
            "EQUALIZATION_VOLTS": 52.4,
            "BOOST_VOLTS": 52.4,
            "FLOAT_VOLTS": 50.8,
            "BOOST_RECONNECT_VOLTS": 50.7,
            "LOW_VOLTS_RECONNECT": 49.9,
            "UNDER_VOLTS_RECOVER": 49.9,
            "UNDER_VOLTS_WARNING": 48.5,
            "LOW_VOLTS_DISCONNECT": 48.5,
            "DISCHARGING_LIMIT_VOLTS": 48.5
        }
    },
    "DEVICE_QUANTITY": 1,
    "DEVICE_FREE_HEAP": 28648,
    "DEVICE_FREE_JSON": 7229,
    "ESP_VCC": 3.297,
    "Wifi_RSSI": -63,
    "sw_version": "2.3.0"
}
```

### What browsers are you seeing the problem on?

Chrome, Safari, Microsoft Edge

softwarecrash, 2024-06-13:
Hello,
Please try to reflash the Firmware over USB.

jurijzahn8019, 2024-06-13:
Hi, 

- i have flashed. didnt work.
- wiped memeory, flashed again with [EPEver2MQTT_d1_mini_2.3.0.bin](https://github.com/softwarecrash/EPEver2MQTT/releases/download/V2.3.0/EPEver2MQTT_d1_mini_2.3.0.bin), reconfigured. 

but didn't work.

I am using https://all-solutions.store/de/shop/epever2mqtt-set/
with wemos d1 pro.

softwarecrash, 2024-06-13:
thanks, so i tested the actual release, that work without any problems.
when you call the setup page can you watch the esp the led light up for 2-5seconds?
i looks the data was not full delivered to the browser, so i dont know why.
can you take a look at the webserial was this output when you call the setup page?

jurijzahn8019, 2024-06-14:
Thanks, i will try another esp.

softwarecrash, 2024-06-14:
Another Thing you can try is to unplug the ESP and Power IT over USB from your PC.
When it Work IT can Happen that the epever doenst provide enugh Power.

jurijzahn8019, 2024-06-14:
Oh, its powered by external usb psu

softwarecrash, 2024-06-14:
Okay, Just try vise versa and Power IT from the epever, Just for confirm.

Is the Display Error from start when the ESP is resteted or comes it up when the User Data is saved?

jurijzahn8019, 2024-06-14:
I have found out, that my mqtt password causes the issue:

password contains: `[a-z][A-Z][0-9]%=`

Seems application doesn't like special characters :)

softwarecrash, 2024-06-14:
ahhhh... sure... the webserver doenst like the % charracter, it is default reserved for the html preprocessor, damn i need to write it down somewhere

jurijzahn8019, 2024-06-14:
Meaning, password must be changed? :)

softwarecrash, 2024-06-14:
That the fields cant have the % character, or i find time to fix that stupid thing

softwarecrash, 2024-06-16:
@jurijzahn8019 
can you test this version?
it will accept % in any input fields, please check the whole webinterface for type errors and other issues
[EPEver2MQTT_d1_mini_2.3.0A5.zip](https://github.com/user-attachments/files/15857509/EPEver2MQTT_d1_mini_2.3.0A5.zip)

jurijzahn8019, 2024-06-17:
Hi @softwarecrash,

I can confirm, that it works, however, I have seen, that you have changed the escape character to make it work. I am not sure, if this helps in the long term, The next guy will use backtick in his password.

On my side, I have created an additional mqtt user with a password adhering to the requirements of your software, and it did not hurt.

On the other hand, I would like to make a suggestion: what about rewriting the frontend as an SPA for example with svelte, and then simply using JSON as the transport format? 

You can see an example implementation here: https://github.com/jurijzahn8019/jt-dpm8600-psu-mqtt

This would eliminate the need to escape and render the app server side.

softwarecrash, 2024-06-17:
@jurijzahn8019 thank you, sure i thinking about to get the data over a json request, but the parts was serverside processed are over the whole web pages and a bit tricky to replace it all, it need a common query for all pages.
so that need rewrite on all projects and actual i work on another thing. so i think i will leave it for now as a little  bug.
btw verry nice code style

closed 2024-08-01 by github-actions[bot]