softwarecrash/EPEver2MQTT #51: [HELP] no data from device

Blackforrestpilot, 2024-09-01:
### A clear and concise description of what the problem is.

Hi,
ich habe einen Epever Tracer 4210an im Betrieb und daran den Adapter für Wlan mit D1 mini (SW Version 2.3.0). Der Laderegler ist nur zum Laden einer 8S LFP verkabelt, Lastausgang wird nicht genutzt, der ESP bekommt die Versorgungsspannung von einem externen 5V Netzteil.
Leider bekomme ich widerholt keine Daten, in der Weboberfläche steht dann "No data from Device 1"
Im Webserial laufen diese Meldungen ein:
[image]

Die Hardware scheint in Ordnung zu sein, da es in 2 von 5 Fällen funktioniert hat, dass der Adapter nach einer SW-Aktualisierung (immer das gleiche BIN-File) Daten bekommen hat. Teilweise nur dann, wenn ich die Zeit aktualisiert habe.

Reproduzierbar ist der Fehler bei mir:
- OTA Update machen --> Daten kommen
- MQTT Refresh (sec) auf 60 ändern --> keine Daten

erneutes Ändern auf 300 Sekunden behebt den Fehler nicht

softwarecrash, 2024-09-02:
Das klingt als wäre die Kommunikation instabil, welche Komponenten hast du den verbaut und wie? Kannst du Fotos davon anhängen?

Blackforrestpilot, 2024-09-03:
[image]
Ich habe heute ein neues Kabel mit dem RJ45 Stecker gemacht, weil ich bisher immer mit dem gleichen Kabel probiert hatte. Leider keine Besserung.
Das ist auch schon der 2. ESP und ein anderes 485 Board.
Bin etwas ratlos, weil ein baugleicher Adapter an dem Laderegler funktioniert hat. Der ist aber jetzt bei einem Kumpel verbaut, also nicht mehr verfügbar.

Blackforrestpilot, 2024-09-04:
Ich habe das mit der 5V ausprobiert, leider weiterhin die gleiche Situation.
seltsam finde ich, dass im Regelfall nach einem "Firmware Update" Daten empfangen werden, aber nur einmalig.

softwarecrash, 2024-09-11:
> Ich habe das mit der 5V ausprobiert, leider weiterhin die gleiche Situation. seltsam finde ich, dass im Regelfall nach einem "Firmware Update" Daten empfangen werden, aber nur einmalig.

das scheint irgendein level problem zu sein, zumal es mit einem anderen setup ja geht

Blackforrestpilot, 2024-10-04:
Da alle meine Versuche keinen Erfolg gebracht haben, bin ich zu einem anderen "Projekt" umgeschwenkt: https://github.com/chickey/RS485-WiFi-EPEver
Die Verkabelung des MAX485-Moduls ist etwas anders, aber hier läuft es mit der selben Hardware stabil.
Etwas komisch.
Kann hier geschlossen werden.

Grüße
Christoph

closed 2024-10-04 by softwarecrash