softwarecrash/EPEver2MQTT #52: [HELP] 226 Read registers for clock failed

kraftandreas, 2024-09-10:
### A clear and concise description of what the problem is.

I followed the instructions entirely including hardware. I keep getting the error above. 

I have a tracer 4210an and already tried reversing A and B on the modbus side which didn’t change anything. 

Can anyone help?

softwarecrash, 2024-09-11:
please show your esp wireing and rs module

kraftandreas, 2024-09-11:
[image]
[image]

this is reversed polarity A/B. 

From D1 Mini to RS485:
TX -> DI
RX -> RO
D1 -> DE+RE (bridged)

5V -> VCC
GND -> GND

from RS485 to RJ45:

VCC -> ORANGE (Pin 2)
GND -> BROWN (Pin 8)
B -> BLUE (Pin 4)
A -> GREEN (Pin 6)

softwarecrash, 2024-09-11:
can your verify that U1 is a orginal chip? some china clones doesnt work well

kraftandreas, 2024-09-11:
[image]

U1 seems original. I ordered specifically the device that was mentioned. 

I will try powering externally and cut the VCC (and possibly GND) line beforehand. 

Thanks for the quick help!

softwarecrash, 2024-09-11:
bevor you change anything, try to set the EP device ID to 1, some devices have a changed ID
https://github.com/softwarecrash/EPEver2MQTT/wiki/Multi-Device-Tutorial
use http://<ESP-IP>/set?devid=1

closed 2024-10-26 by github-actions[bot]