softwarecrash/EPEver2MQTT #54: Date and Time setting and MT50 parallel EPEver2MQTT error

rt1970, 2024-09-23:
### A clear and concise description of what the problem is.

Hello!
1. Can Date and Time-Setting configured with SNTP?
2. If i have MT50 and EPEver2MQTT both, then any measures are wrong.
My MPPT: EPEver Tracer 5210 AN

rt1970, 2024-09-23:
do you have a solution for MT50? Both are on RS485 RJ45 connector. EPEver has 2 RJ45 connectors.
I have disconnected the 2x 100 Ohm but no changes...

softwarecrash, 2024-09-23:
Normal ESP and mt50 can exsist in Same wireing.
But the ESP is a master and the mt50 too, so i Made a little Timing Trick that both can Work.
So when Set Tine Not Work try repeatly it, or disconnect the mt50 for this time.
Normaly only one master can exist in the Network and Control one or more slaves

rt1970, 2024-09-23:
Problems are without time setting.
2x Master... I think thats possible...
Can i do EPEver2MQTT configure as slave?

softwarecrash, 2024-09-23:
No, ESP and/or mt50 must be master
That is was master and Slave does, the master ask the Slave, and the Slave answer it.
A Slave cant ask a Slave and vise versa.

rt1970, 2024-09-23:
No solution?
As example: i have sometime 6315662A oder 23342323V or so. Other numbers are sometimes also crazy

softwarecrash, 2024-09-23:
There is No solution, thats where modbus Work.
When the wrong numbers gone when disconnect the mt50 then my Trick doenst Work for your Setup.
When the wrong numbers stay, the Numbers come from the device itself, every Data Message have a checksum so that the Chance of wrong Data is minimal

rt1970, 2024-09-24:
Ok. Thank you for your explanation! Best way for is then ESP with OLED.

softwarecrash, 2024-09-24:
> Ok. Thank you for your explanation! Best way for is then ESP with OLED.
> 
> Device Time: after one Day is clock 1h to late...

thats a lot of drift... some devices have minutes over month

closed 2024-11-09 by github-actions[bot]