softwarecrash/EPEver2MQTT #60: [HELP] change Baudrate

andre1210, 2024-12-29:
### A clear and concise description of what the problem is.

Is it possible to configure a different Baudrate?
My Epever tracer is running since 3 years without issue using solaranzeige and an usb ftty adapter. Due to issues, the baudrate was set to 19200.
Now I want to use the wireless solution in order to relocate the Servers.
The Epever Software supports only Windows and currently I don‘t have a Windows PC near any more.
Are there any other possibilities to change Baudrate configuration. Either on esp side or on Epever side?
Thank you

softwarecrash, 2024-12-30:
some devices let you change the baud rate over the display, elsewhere i dont have any idea to change it without pc software.
or compile the project by yourself with the modified baudrate

miky2k, 2025-01-12:
>  The Epever Software supports only Windows and currently I don‘t have a Windows PC near any more. 
You dont't need to have pc near , you can use a tcp/serial bridge (tasmota/esplink ecc ecc) and use tcp2com under windows to create virtual network serial port.
I would adopt this project but i need to configure epever charge controller with epever software so i'm stuck with other firmwares that give access to rs485 serial but thorough tcp.

andre1210, 2025-01-12:
I found an old Notebook where I could use the Epever software to change the baud rate.
Now EPEver2Mqtt works fine.
Maybe an option to configure the baudrate would be nice ;-)
Thank you for this project.

closed 2025-01-12 by andre1210

miky2k, 2025-01-12:
> I found an old Notebook where I could use the Epever software to change the baud rate. Now EPEver2Mqtt works fine. Maybe an option to configure the baudrate would be nice 

Ok you solve YOUR problem  but the feature request is not over , strange behavior.