softwarecrash/EPEver2MQTT #62: [FEATURE] Setting time over MQTT

BerziOnline, 2025-03-08:
### Is your feature request related to a problem? Please describe

When the device losing it's power the time and date of the EPEver is being set back to default.
You have to find out and then set it back manually over the browser frontend, so you have no chance to correct the time automatically over your smart home.

### Describe the solution you'd like

It would be a charme if the MQTT datapoint of the device time would be writeable, so one could have an automation what directly corrects the time over MQTT, when the time is wrong, instead just giving a note that you have to correct it manually.

### Describe alternatives you've considered

NTP would be even nicer, but at least writing the time over MQTT would be an improvement.

softwarecrash, 2025-03-09:
its possible, but what time format use? it exsist so many datetime format strings, and pay attention to every variant and typo that its bit hard

NTP is nice, but same thing, a lot of time zones and daylight saving zones what need to set correctly.

any suggestions to realize it?

BerziOnline, 2025-03-09:
Thanks for your reply! I understand the concerns about handling different time formats and potential issues with time zones. To keep things simple and reliable, I’d suggest using a Unix timestamp with 13 digits (milliseconds since epoch) for setting the time via MQTT.

This approach has several advantages:

Universal format – avoids issues with different datetime formats.

Timezone-independent – the timestamp is always in UTC, and any system reading it can convert it to the correct local time.

Simple to implement – most programming languages and systems support Unix timestamps out of the box.

Regarding NTP, I think time zones shouldn't be a major obstacle. NTP always provides UTC time, and it’s up to the device to apply the correct local time offset. If the EPEver device doesn’t support time zone handling internally, I understand why that could be a concern. But if it does, then NTP would be the most robust long-term solution.

That said, having the ability to set the time via MQTT with a Unix timestamp would already be a huge improvement, as it allows automation systems to correct the time when needed.

Would this be a feasible solution?

softwarecrash, 2025-03-09:
unix time is a usable format.

the epever doenst care about timezones and whatever, it takes the time you give it, thats the problem with timezones, it need to adjust in the esp bevore use, same for unix time. 

you can play arround with `<esp-ip>/set?datetime=`
its undocumented, but a short help wit a webhook.

more about this here
https://github.com/softwarecrash/EPEver2MQTT/blob/58ba7738ac20bc2bd8db9d783fcf765465604665/src/webpages/HTML_SETTINGS.html#L95

let me check the next week that made a new mqtt point that is subscribed and accept unix timecode and transfer it to all devices.
badly i doent have a epever at the moment for testing here

BerziOnline, 2025-03-10:
Yesss, I know what you're complaining about because of the time behaviour of the charge controller itself. It does not work with a time zone based time stamp, it's really weird. I was struggling with it when I read out the time stamp out of the topic DEVICE_TIME, what is a 10 digit unix time. Firt it is little bit different because it's not in 13 digits, but okay. But the main issue here is, that it is using a timestamp including timezone calculation, but saving it as UTC. So when you work on your broker with that timezone, it's used as UTC, then it calculates your timezone offset to it and of course it's wrong then. For the moment it looks easy solveable by calculating directly with a new correction-offset, but then you have to be aware between summer and winter time again and you're going to start writing your own timezone offset on another system :D 

For the moment what I do in my broker is using the java function Date.getTimezoneOffset() and reduce that from the given epever time (DEVICE_TIME), so I know I have the UTC for further time calculations on my broker. Just to let you know, I am understanding what logical time issues the charge controller brings in.

A good hint with the webhook, thanks a lot! Would be a solution in between. I tried it manually, but I had no success - but I think just because I'm to stupid what kind of dateString format he's expecting. Do you know that? I tried the URL with a 10 digit unix time and a 13 digit unix time to see if the time in the charge controller is changing.

All I get is a "message received" after the webhook - so the webhook itself should work. But the time is not being adjusted, so I think I am just using the wrong format.

Do you know how an example dateString for the URL could look like?

While you're not having an epever available on your side, I can test things for you. 

Thanks a lot for your help!!

softwarecrash, 2025-03-10:
for the webhook, i have do wired things here for get it working long ago, i think you must send it as timedate format without special character, like `<esp-ip>/set?datetime=250310100326`  so the string format is YYMMDDhhmmss

yes the unix time you get over mqtt is wrong based on the timezone, so the esp doesnt know your timezone to calculate it right, thats the big mistake, the epever doenst care about the timezone, and the esp doenst know it so that cant go right ;)
i have chosen the unix format for mqtt because the mqtt client library have some issues with strings that change often.
i dont have invest more time to the time thing because i think its not nescary, so now i will thin/work again on it.

i have a bit strange idea how to get NTP working with it, need some testing

BerziOnline, 2025-03-10:
Yeah! This string is working like a charme - thanks a lot! I think with the webhook at least there is a workaround.

Thank you for working on the NTP solution!

softwarecrash, 2025-03-11:
@BerziOnline here a first version with NTP
you need to go to the setup and click on the link on timezone field, search the timezone code and put it in the textfield, after save, the esp try every 15 min to update the time on the devices, please try also different timezones and check the epever device time that it set correct

[EPEver2MQTT_d1_mini_2.3.1N1pack.bin.zip](https://github.com/user-attachments/files/19195977/EPEver2MQTT_d1_mini_2.3.1N1pack.bin.zip)

BerziOnline, 2025-03-12:
Awesome improvement! It's working like a charme. So at least I tested changing some timezones and after rebooting, without setting any manually the time has been shown correctly.

I tested i. e. Los Angeles with "PST8PDT,M3.2.0,M11.1.0" and then changed back to Berlin with "CET-1CEST,M3.5.0,M10.5.0/3". 
Also did some others, but these both I checked for sure and the time has been correct after rebooting.

I like the way you linked the timezones. Maybe it would be helpful to give a hint what kind of field you have to copy? I think for someone who don't know this issue here they could think pasting "Europe/Berlin" instead of "CET-1CEST,M3.5.0,M10.5.0/3" maybe? But not very important, I think one could try & error.

What could be also interesting is setting up the ntp domain or IP on your own. Right now it's hardcoded to "pool.ntp.org", but I think there are some people want to use such IoT-Devices as locally as possible. 

Anyway - the function is working and thanks a lot for this fast and motivated work on that!!

softwarecrash, 2025-03-12:
> Awesome improvement! It's working like a charme. So at least I tested changing some timezones and after rebooting, without setting any manually the time has been shown correctly.
>
> I tested i. e. Los [...]
great, timezones work correct.
for testing the update interval is set to 15min, default by SDK is 60min update, is this enugh or need shorter?

> I like the way you linked the timezones. Maybe it would be helpful to give a hint what kind of field you have to copy? I think for someone who don't know this issue here they could think pasting [...]
sure, but the hint field is limited, i think about it, or any idea for it?

> What could be also interesting is setting up the ntp domain or IP on your own. Right now it's hardcoded to "pool.ntp.org", but I think there are some people want to use such IoT-Devices as locally as possible.
> 
thats comming next, the N1 is only a first test of working, another input field with prefilled pool.ntp.org but changeable is okay.

> Anyway - the function is working and thanks a lot for this fast and motivated work on that!!

need bit more work, but will remove the unix timestamp from device to mqtt and replace with humand readable time string.
so i will post you the next test candidate here when it finish some little things

BerziOnline, 2025-03-12:
> for testing the update interval is set to 15min, default by SDK is 60min update, is this enugh or need shorter?

I would also say 60min should be enough, to keep overhead as minimal as possible?

> sure, but the hint field is limited, i think about it, or any idea for it?

Hmm... yes I understand. Also because you wanna explain, that nothing means no NTP active. What if you type in something like: `use 'CET-1CEST,M3.5.0,M10.5.0/3' for Europe/Berlin` and better ignore explanation about the "nothing" status? Or if it's not to long `Empty = disable sync; E.g. for Europe/Berlin = CET-1CEST,M3.5.0,M10.5.0/3`. 
Most stable option would be to use a dropdown-field, but I think this is a mess for you to setup.

> thats comming next, the N1 is only a first test of working, another input field with prefilled pool.ntp.org but changeable is okay.

Sounds great! Please keep in mind that maybe some people want to use a domain and some want (or have to) use an IP address.

> need bit more work, but will remove the unix timestamp from device to mqtt and replace with humand readable time string.

Are you sure it's a good idea to send times as string? Most devices I know sending timestamps by unix-timestamps, because most systems can directly work with that format. Sending time in string-format forces the receiver to split and re-calculate everything to go on in further work with it. If you really plan to format the time in a better way I would suggest to calculate the timestamp to UTC unix time stamp with 13 digits. This is the best possible source format for every other system. They can easily display that human readable in every liked timezone later on, use it for calculations or time based comparisations, and so on. What I know is in java there's a function for the timezone giving you just the offset back (Date.getTimezoneOffset()). If you're able to extract the timzone offset from the actual setting as number then you can just reduce or add that to your actual unix timestamp and it is UTC. This would be my idea - but please don't focus on me and my ideas, just trying to brainstorm as help. For me and my case you did more than needed (thanks a lot for that!). I don't want to get annoying with more ideas - just did that, while you told a little bit from your plans.

> so i will post you the next test candidate here when it finish some little things

No problem - I will test and give you feedback.

softwarecrash, 2025-03-12:
@BerziOnline 
here are a more finished version.
needed some upstream to actual librarys, so can you please check that Load switch over mqtt work?

[EPEver2MQTT_d1_mini_2.3.1N2_pack.bin.zip](https://github.com/user-attachments/files/19210509/EPEver2MQTT_d1_mini_2.3.1N2_pack.bin.zip)

BerziOnline, 2025-03-17:
Thank you for the update.
I have checked the 2.4.0 version.

NTP and NTP-Server address is working. Also the hint in the empty field is helpful I think.
I also tested your wish about the Load-Button over MQTT. The result is:

- Load-Button over Website is working
- MQTT-Object of the Button is correct read, but changes by MQTT has no effect

If you change Load over the Website in the next period is correct sent to the MQTT broker. But if you change it within the broker it is ignored by the EPEver and within the next period the EPEver is correcting the (now wrong) status within the broker back to normal. So changing the load button over website is working, but not over MQTT. But the status from the load button is synced from EPEver to the broker.

softwarecrash, 2025-03-17:
> Thank you for the update. I have checked the 2.4.0 version.
>
> NTP and NTP-Server address is working. Also the hint in the empty field is helpful I think. I also tested your wish about the Load-Button [...]

short question, have you tried it over the right mqtt control?
https://github.com/softwarecrash/EPEver2MQTT/wiki/Output-Control
the load state have a read and seperate set path

BerziOnline, 2025-03-17:
> short question, have you tried it over the right mqtt control? https://github.com/softwarecrash/EPEver2MQTT/wiki/Output-Control the load state have a read and seperate set path

Alright! With that object it's working! Tested it right now.

softwarecrash, 2025-03-17:
> > short question, have you tried it over the right mqtt control? https://github.com/softwarecrash/EPEver2MQTT/wiki/Output-Control the load state have a read and seperate set path
> 
> Alright! With that object it's working! Tested it right now.

thank you for testing :)