softwarecrash/EPEver2MQTT #65: [Bug]: MQTT-URL too long for the input mask (Azure Portal and HiveMQ)

Jan08, 2025-03-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

I want to use a MQTT broker like Azure or HiveMQ as I have already connected other sensors to this system.
For this I have to enter the rather long MQTT URL: e18111546485492c80c0ia59c627f3eb.s1.eu.hivemq.cloud But when I do this it truncates to e18111546485492c80c0ia59c627f3eb.s1 I guess the fix is quite simple by increasing the number of characters from 35 to 70 or so.

### Screenshots / Fotos

[image]

[image]

### Steps To Reproduce

Got to Edit Configuration try to enter the URL

### Version

1.x.x and above

### Relevant livejson output

```json
{"DEVICE_NAME":"EPEver2MQTT","EP_1":{"LiveData":{"CONNECTION":0,"DEVICE_NUM":"1","DEVICE_TIME":1742066752,"DEVICE_TEMP":8.66,"SOLAR_V":0.09,"SOLAR_A":0,"SOLAR_W":0,"BATT_SOC":68,"BATT_V":13.17,"BATT_A":-1.06,"BATT_W":-13.1,"BATT_STATE":"Normal","BATT_TEMP":7.34,"BATT_TEMP_STATE":"Normal","LOAD_V":13.17,"LOAD_A":1.11,"LOAD_W":14.61,"LOAD_STATE":true,"CHARGER_STATE":"Normal","CHARGER_MODE":"Off"},"StatsData":{"SOLAR_MAX":89.85,"SOLAR_MIN":0.07,"BATT_MAX":14.7,"BATT_MIN":13.09,"CONS_DAY":0.06,"CONS_MON":1.05,"CONS_YEAR":4.77,"CONS_TOT":23.24,"GEN_DAY":0.15,"GEN_MON":4.6,"GEN_YEAR":9.49,"GEN_TOT":56.82,"CO2_REDUCTION":0.05},"DeviceData":{"BATTERY_TYPE":"User","BATTERY_CAPACITY":300,"TEMPERATURE_COMPENSATION":0,"HIGH_VOLT_DISCONNECT":14.8,"CHARGING_LIMIT_VOLTS":14.2,"OVER_VOLTS_RECONNECT":14.2,"EQUALIZATION_VOLTS":14.2,"BOOST_VOLTS":14.2,"FLOAT_VOLTS":13.8,"BOOST_RECONNECT_VOLTS":13.2,"LOW_VOLTS_RECONNECT":12.6,"UNDER_VOLTS_RECOVER":12.2,"UNDER_VOLTS_WARNING":12,"LOW_VOLTS_DISCONNECT":11.8,"DISCHARGING_LIMIT_VOLTS":11}},"DEVICE_QUANTITY":1,"DEVICE_FREE_HEAP":36376,"ESP_VCC":3.331,"Runtime":338,"Wifi_RSSI":-72,"sw_version":"2.4.0"}
```

### What browsers are you seeing the problem on?

Firefox

softwarecrash, 2025-03-15:
the memory space is limited to 40 chars, try a url shortener

Jan08, 2025-03-16:
Thanks for the quick replay. I see that you have increased the number of characters to 127, but it still does not work. There could be several reasons for this. On the one hand I'm trying to connect to a web MQTT server, so I'm not sure if the device is programmt to connect to the internet or has it just excess to the lan only?
But more likely is that the string is not processed after the user input. It seems to me that it is truncated to a length of 40 characters.

[image]

softwarecrash, 2025-03-16:
> Thanks for the quick replay. I see that you have increased the number of characters to 127, but it still does not work. There could be several reasons for this. On the one hand I'm trying to connect [...]

yes, i forgot to change that
try this

[EPEver2MQTT_d1_mini_2.4.0s1.zip](https://github.com/user-attachments/files/19270951/EPEver2MQTT_d1_mini_2.4.0s1.zip)

remember that there is now way to support secure mqtt, your web mqtt server must support insecure connection

Jan08, 2025-03-16:
Hey, I got platformIO running and got it running after some time ;)

So following bug fix:

1.  The sanitychacke in Setting.h was overwriting any string longer then 40 chars:

  void sanitycheck()
  {
    if (strlen(data.deviceName) == 0 || strlen(data.deviceName) >= 40)
    {
      strcpy(data.deviceName, "EPEver2MQTT");
    }
    if (strlen(data.mqttServer) == 0 || strlen(data.mqttServer) >= 40 )      # change to 128
    {
      strcpy(data.mqttServer, "");

Aside from that I needed to change the Client form WiFiClient to WiFiClientSecure and set it to   
client.setInsecure(); befor the mqttclient.setServer(_settings.data.mqttServer, _settings.data.mqttPort);

This did the trick to connect to a 8883 Port...
Do I know what I did by this last change? No ;) I hope it wont cause any problems

softwarecrash, 2025-03-16:
ah sanity another one that i forgot... thanks

so when you can test with the wificlient changes that it work with secure and insecure connections i can apply it to the release

Jan08, 2025-03-18:
Hey, sofar the bord is running flawless, but I can't test if WiFiClientSecure and client.setInsecure(); works with a lokal MQTT server (192.168.1.1.xxx) as well, as I dont have a running MQTT-Server and as the device is on my countryside cottage.

softwarecrash, 2025-03-19:
Just testet with the Changes, badly ist Crashes  when trying tonconnect to a Classic insecure mqtt Server, i think it need a lot more Changes to Work with both

Jan08, 2025-03-19:
That's sad 🫤
But at least here is a solution documented for anyone who stumbles about the same issue.
I guess you can close the Issue as solved.
Thank you for your help and quick replies!
You did a great job on this project 👍🏻

closed 2025-05-03 by github-actions[bot]