softwarecrash/EPEver2MQTT #66: [Bug]: Wrong readings from Epever

MStapelfeldt, 2025-04-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Hello, sometimes I have a strange behaivor in `sensor.epever2mqtt_ep_1_gen_tot` statistics from HA: I see negativ readings, which can't be. The sensor reports a total increasing value and over a regular day, I see a reading between 0,00 and 0,05 every 5 Mintutes. From time to time (maybe every 2-3 Days) I have one value which shows - 1,00 - -8,00 which is not possible. It would be great to filter out this wrong readings.

### Screenshots / Fotos

[image]
[image]
[image]

### Steps To Reproduce

Can't reproduce, it's random

### Version

0.2.4

### Relevant livejson output

```json
live json can be send later when back home, but you can't see the issue in live system
```

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2025-04-14:
can you log it direct on the mqtt?

MStapelfeldt, 2025-04-14:
It only happens once a day, so it will be a long 24h log, but I can try if you could point me to where to activate it. 😅

softwarecrash, 2025-04-14:
just checked the code, the gen_total datapoint is a uint32_t datatype, the u means unsigned, so they never can have a negative value.
i think the fault comes from the HA or its calculations of the total energy or logging, so when it comes only once a day, this is the point where HA think its time for a new day of calculation  and miscalculate it.

so there on  the code self i cant change anything to correct this behavior

softwarecrash, 2025-04-14:
badly there is no raw data.
take a look at grafana and influxdb, i cant help you by this i dont use HA or its envroiment

MStapelfeldt, 2025-04-14:
Ok, I'll check tomorrow at work. Bigger PC and screens if I can find the entity values in DB after I find a faulty reading. Already cleaned them from the weekend today. 😅 I'll come back to you if I find some data for you.

softwarecrash, 2025-04-14:
great.
sidenote, its nearly impossible that the esp send you wrong data, the data from the epever recived was checked with crc, the crc is not unfailable but the chance that wrong data calculated at right crc is 1/10.00000... and a lot more numbers

MStapelfeldt, 2025-04-15:
Hey, I got a small missreading today at 12:22:57 and checked the debug log:
```
2025-04-15 12:22:27.078 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on EPEver/EP_1/StatsData/GEN_TOT (qos=0): b'783.13'
2025-04-15 12:22:57.974 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on EPEver/EP_1/StatsData/GEN_TOT (qos=0): b'782.49'
2025-04-15 12:23:30.391 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on EPEver/EP_1/StatsData/GEN_TOT (qos=0): b'782.49'
2025-04-15 12:24:01.450 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on EPEver/EP_1/StatsData/GEN_TOT (qos=0): b'782.5'
```
Strange, but as you see: Total 783,13 became 782,49, so generated solar energie was -0,64, which can't be. After that, it continued to count upwards... I'm realy not sure if this is an epever2mqqt problem when looking at these numbers. I think there is something wrong with my eperver tracer...

Here is also the full log with all messages, but the errors are not related to epever2mqqt.
[mqqt epever.log](https://github.com/user-attachments/files/19760679/mqqt.epever.log)

softwarecrash, 2025-04-15:
Im Sure that comes from the epever device self.
Have you tried tonactivate the ntp sync?
The epever have a internal clock thats Not accurate, evt miscalculate it deepens on this, when ntp is in, Switch it off

closed 2025-05-31 by github-actions[bot]