softwarecrash/EPEver2MQTT #67: [Bug]: Cant change LOAD_STATE using MQTT

sdalu, 2025-05-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Device is configure to use `MQTT Json Style`, so I'm pushing the data as `{"EP_1":{"LiveData":{"LOAD_STATE":false}}}`.

But it doesn't seem to change the load state, as show after triggering an mqtt request

~~~
mosquitto_pub ${mqtt_cfg} -t epever2mqtt/DATA -m '{"EP_1":{"LiveData":{"LOAD_STATE":false}}}'
mosquitto_pub ${mqtt_cfg} -t epever2mqtt/trigger -m 1
mosquitto_pub ${mqtt_cfg} -t epever2mqtt/DATA -m '{"EP_1":{"LiveData":{"LOAD_STATE":false}}}'
mosquitto_pub ${mqtt_cfg} -t epever2mqtt/trigger -m 1
~~~

### Screenshots / Fotos

~~~
epever2mqtt/DATA {"DEVICE_NAME":"EPEver2MQTT","Wifi_RSSI":-80,"EP_1":{"LiveData":{"CONNECTION":0,"DEVICE_NUM":"1","DEVICE_TIME":1728585282,"DEVICE_TEMP":18.93,"SOLAR_V":14.6,"SOLAR_A":0.05,"SOLAR_W":0.79,"BATT_SOC":81,"BATT_V":13.32,"BATT_A":-0.27,"BATT_W":-2.66,"BATT_STATE":"Normal","BATT_TEMP":17.23,"BATT_TEMP_STATE":"Normal","LOAD_V":13.32,"LOAD_A":0.26,"LOAD_W":3.46,"LOAD_STATE":true,"CHARGER_STATE":"Normal","CHARGER_MODE":"Boost"},"StatsData":{"SOLAR_MAX":24.6,"SOLAR_MIN":0,"BATT_MAX":21.08,"BATT_MIN":12,"CONS_DAY":0,"CONS_MON":0,"CONS_YEAR":2.57,"CONS_TOT":6.65,"GEN_DAY":0.05,"GEN_MON":0.08,"GEN_YEAR":3.51,"GEN_TOT":6.69,"CO2_REDUCTION":0},"DeviceData":{"BATTERY_TYPE":"User","BATTERY_CAPACITY":200,"TEMPERATURE_COMPENSATION":3,"HIGH_VOLT_DISCONNECT":14.8,"CHARGING_LIMIT_VOLTS":14.6,"OVER_VOLTS_RECONNECT":14.6,"EQUALIZATION_VOLTS":14.5,"BOOST_VOLTS":14.5,"FLOAT_VOLTS":13.8,"BOOST_RECONNECT_VOLTS":13.2,"LOW_VOLTS_RECONNECT":12.8,"UNDER_VOLTS_RECOVER":12.2,"UNDER_VOLTS_WARNING":12,"LOW_VOLTS_DISCONNECT":11.1,"DISCHARGING_LIMIT_VOLTS":11}},"DEVICE_QUANTITY":1,"DEVICE_FREE_HEAP":34872,"ESP_VCC":3.293,"Runtime":32,"sw_version":"2.5.0"}
epever2mqtt/DATA {"EP_1":{"LiveData":{"LOAD_STATE":false}}}
epever2mqtt/trigger 1
epever2mqtt/Alive true
epever2mqtt/Wifi_RSSI -80
epever2mqtt/DATA {"DEVICE_NAME":"EPEver2MQTT","Wifi_RSSI":-80,"EP_1":{"LiveData":{"CONNECTION":0,"DEVICE_NUM":"1","DEVICE_TIME":1728585296,"DEVICE_TEMP":18.92,"SOLAR_V":14.6,"SOLAR_A":0.06,"SOLAR_W":0.93,"BATT_SOC":82,"BATT_V":13.32,"BATT_A":-0.2,"BATT_W":-2.66,"BATT_STATE":"Normal","BATT_TEMP":17.23,"BATT_TEMP_STATE":"Normal","LOAD_V":13.32,"LOAD_A":0.31,"LOAD_W":4.12,"LOAD_STATE":true,"CHARGER_STATE":"Normal","CHARGER_MODE":"Boost"},"StatsData":{"SOLAR_MAX":24.6,"SOLAR_MIN":0,"BATT_MAX":21.08,"BATT_MIN":12,"CONS_DAY":0,"CONS_MON":0,"CONS_YEAR":2.57,"CONS_TOT":6.65,"GEN_DAY":0.05,"GEN_MON":0.08,"GEN_YEAR":3.51,"GEN_TOT":6.69,"CO2_REDUCTION":0},"DeviceData":{"BATTERY_TYPE":"User","BATTERY_CAPACITY":200,"TEMPERATURE_COMPENSATION":3,"HIGH_VOLT_DISCONNECT":14.8,"CHARGING_LIMIT_VOLTS":14.6,"OVER_VOLTS_RECONNECT":14.6,"EQUALIZATION_VOLTS":14.5,"BOOST_VOLTS":14.5,"FLOAT_VOLTS":13.8,"BOOST_RECONNECT_VOLTS":13.2,"LOW_VOLTS_RECONNECT":12.8,"UNDER_VOLTS_RECOVER":12.2,"UNDER_VOLTS_WARNING":12,"LOW_VOLTS_DISCONNECT":11.1,"DISCHARGING_LIMIT_VOLTS":11}},"DEVICE_QUANTITY":1,"DEVICE_FREE_HEAP":34872,"ESP_VCC":3.293,"Runtime":46,"sw_version":"2.5.0"}
epever2mqtt/DATA {"EP_1":{"LiveData":{"LOAD_STATE":false}}}
epever2mqtt/trigger 1
epever2mqtt/Alive true
epever2mqtt/Wifi_RSSI -80
epever2mqtt/DATA {"DEVICE_NAME":"EPEver2MQTT","Wifi_RSSI":-80,"EP_1":{"LiveData":{"CONNECTION":0,"DEVICE_NUM":"1","DEVICE_TIME":1728585324,"DEVICE_TEMP":18.92,"SOLAR_V":14.57,"SOLAR_A":0.05,"SOLAR_W":0.79,"BATT_SOC":81,"BATT_V":13.31,"BATT_A":-0.27,"BATT_W":-2.66,"BATT_STATE":"Normal","BATT_TEMP":17.2,"BATT_TEMP_STATE":"Normal","LOAD_V":13.31,"LOAD_A":0.28,"LOAD_W":3.72,"LOAD_STATE":true,"CHARGER_STATE":"Normal","CHARGER_MODE":"Boost"},"StatsData":{"SOLAR_MAX":24.6,"SOLAR_MIN":0,"BATT_MAX":21.08,"BATT_MIN":12,"CONS_DAY":0,"CONS_MON":0,"CONS_YEAR":2.57,"CONS_TOT":6.65,"GEN_DAY":0.05,"GEN_MON":0.08,"GEN_YEAR":3.51,"GEN_TOT":6.69,"CO2_REDUCTION":0},"DeviceData":{"BATTERY_TYPE":"User","BATTERY_CAPACITY":200,"TEMPERATURE_COMPENSATION":3,"HIGH_VOLT_DISCONNECT":14.8,"CHARGING_LIMIT_VOLTS":14.6,"OVER_VOLTS_RECONNECT":14.6,"EQUALIZATION_VOLTS":14.5,"BOOST_VOLTS":14.5,"FLOAT_VOLTS":13.8,"BOOST_RECONNECT_VOLTS":13.2,"LOW_VOLTS_RECONNECT":12.8,"UNDER_VOLTS_RECOVER":12.2,"UNDER_VOLTS_WARNING":12,"LOW_VOLTS_DISCONNECT":11.1,"DISCHARGING_LIMIT_VOLTS":11}},"DEVICE_QUANTITY":1,"DEVICE_FREE_HEAP":34264,"ESP_VCC":3.293,"Runtime":74,"sw_version":"2.5.0"}
~~~

### Version

1.x.x and above

### Relevant livejson output

```json
{"DEVICE_NAME":"EPEver2MQTT","Wifi_RSSI":-80,"EP_1":{"LiveData":{"CONNECTION":0,"DEVICE_NUM":"1","DEVICE_TIME":1728585738,"DEVICE_TEMP":18.89,"SOLAR_V":14.56,"SOLAR_A":0.06,"SOLAR_W":0.92,"BATT_SOC":80,"BATT_V":13.27,"BATT_A":-0.24,"BATT_W":-2.64,"BATT_STATE":"Normal","BATT_TEMP":17.19,"BATT_TEMP_STATE":"Normal","LOAD_V":13.27,"LOAD_A":0.26,"LOAD_W":3.45,"LOAD_STATE":true,"CHARGER_STATE":"Normal","CHARGER_MODE":"Boost"},"StatsData":{"SOLAR_MAX":24.6,"SOLAR_MIN":0,"BATT_MAX":21.08,"BATT_MIN":12,"CONS_DAY":0,"CONS_MON":0,"CONS_YEAR":2.57,"CONS_TOT":6.65,"GEN_DAY":0.05,"GEN_MON":0.08,"GEN_YEAR":3.51,"GEN_TOT":6.69,"CO2_REDUCTION":0},"DeviceData":{"BATTERY_TYPE":"User","BATTERY_CAPACITY":200,"TEMPERATURE_COMPENSATION":3,"HIGH_VOLT_DISCONNECT":14.8,"CHARGING_LIMIT_VOLTS":14.6,"OVER_VOLTS_RECONNECT":14.6,"EQUALIZATION_VOLTS":14.5,"BOOST_VOLTS":14.5,"FLOAT_VOLTS":13.8,"BOOST_RECONNECT_VOLTS":13.2,"LOW_VOLTS_RECONNECT":12.8,"UNDER_VOLTS_RECOVER":12.2,"UNDER_VOLTS_WARNING":12,"LOW_VOLTS_DISCONNECT":11.1,"DISCHARGING_LIMIT_VOLTS":11}},"DEVICE_QUANTITY":1,"DEVICE_FREE_HEAP":35208,"ESP_VCC":3.293,"Runtime":488,"sw_version":"2.5.0"}
```

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

softwarecrash, 2025-05-08:
is this problem coming up with 2.5.0? try downgrade?

TronTronix, 2025-06-14:
Bei der Version 2.5.1 ist es auf jeden fall noch relewant das Problem.

softwarecrash, 2025-06-14:
> Bei der Version 2.5.1 ist es auf jeden fall noch relewant das Problem.

Wurde getestet und funktioniert in der Umgebung problemlos. Da bedarf es mehr Informationen, da der TE Ersteller sich mal wieder nicht weiter geäußert hat...

TronTronix, 2025-06-15:
Also aufgefallen ist es mir schon bei der Version 2.5.0, ich komme von einer älteren Version EPEver2MQTT 2.3.1 sofern ich wieder auf der Version zurückgehe läuft es oder besser geschrieben ist der Ausgang wieder für mein Programm ansprechbar.

Oder hat sich was in der Ansteuerung geändert?

    msg.payload = { "EP_1": { "LiveData": { "LOAD_STATE": false } } };
    msg.payload = { "EP_1": { "LiveData": { "LOAD_STATE": true } } };

[image]

softwarecrash, 2025-06-15:
probier mal diese version bitte

[EPEver2MQTT_d1_mini_2.5.2A1.zip](https://github.com/user-attachments/files/20745411/EPEver2MQTT_d1_mini_2.5.2A1.zip)

TronTronix, 2025-06-15:
mit der EPEver2MQTT 2.3.1 mit JSON-Modus ist es ja bei EPEver/DATA mit msg.payload = { "EP_1": { "LiveData": { "LOAD_STATE": true } } }; das geht mit der neuen nicht und EPEver/EP_1/DATA habe ich auch probiert ohne erflog. 

[image]

TronTronix, 2025-06-15:
ja, die neue EPEver2MQTT 2.5.2A1 ist drauf auf mein ESP Wemos D1 Mini Pro.

softwarecrash, 2025-06-15:
> ja das ist es (entweder hast du es auf json gestellt im setup, dann geht nur json, oder es gehen nur normale datenpunkte, beides gleichzeitig nicht) Danke, kannst du den Quelcood EPEver2MQTT 2.5.2A1 mir geben ich habe noch einige erweiterungen mit eingebaut.

der commt ins main repo wenn der fehler behoben ist, oder wolltest du mir damit sagen das es jetzt geht?

closed 2025-06-15 by softwarecrash