softwarecrash/EPEver2MQTT #68: [Bug]: WiFi Strength indicator is not displayed

pillepallepeng, 2025-05-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Used Hardware?

Other (please write your hardware in 'What happened?')

### What happened?

- WiFi Strength indicator is not displayed
- Hardware: Epever2MQTT NG
- Software Version: 2.5.0 (!v2.5.0 could not be selected in this form!)

### Screenshots / Fotos

[image]

### Steps To Reproduce

1. open epever2mqtt main page
2. WiFi Strength indicator is not displayed

### Version

0.2.4

### Relevant livejson output

```json
{"DEVICE_NAME":"EPEver2MQTT","Wifi_RSSI":-66,"EP_1":{"LiveData":{"CONNECTION":0,"DEVICE_NUM":"1","DEVICE_TIME":1747219402,"DEVICE_TEMP":24.33,"SOLAR_V":69.77,"SOLAR_A":1.84,"SOLAR_W":129.06,"BATT_SOC":60,"BATT_V":50.22,"BATT_A":2.58,"BATT_W":125.5,"BATT_STATE":"Normal","BATT_TEMP":19.17,"BATT_TEMP_STATE":"Normal","LOAD_V":50.22,"LOAD_A":0.02,"LOAD_W":1,"LOAD_STATE":true,"CHARGER_STATE":"Normal","CHARGER_MODE":"Boost"},"StatsData":{"SOLAR_MAX":77.71,"SOLAR_MIN":0.84,"BATT_MAX":52.17,"BATT_MIN":50.14,"CONS_DAY":0.01,"CONS_MON":0.11,"CONS_YEAR":0.14,"CONS_TOT":0.14,"GEN_DAY":0.17,"GEN_MON":37.62,"GEN_YEAR":129.27,"GEN_TOT":129.28,"CO2_REDUCTION":0.12},"DeviceData":{"BATTERY_TYPE":"User","BATTERY_CAPACITY":88,"TEMPERATURE_COMPENSATION":0,"HIGH_VOLT_DISCONNECT":59.2,"CHARGING_LIMIT_VOLTS":57.2,"OVER_VOLTS_RECONNECT":57.4,"EQUALIZATION_VOLTS":57,"BOOST_VOLTS":56.8,"FLOAT_VOLTS":54,"BOOST_RECONNECT_VOLTS":52.8,"LOW_VOLTS_RECONNECT":51.2,"UNDER_VOLTS_RECOVER":48.8,"UNDER_VOLTS_WARNING":48,"LOW_VOLTS_DISCONNECT":44.4,"DISCHARGING_LIMIT_VOLTS":44}},"DEVICE_QUANTITY":1,"DEVICE_FREE_HEAP":23584,"ESP_VCC":3.81,"Runtime":5538,"sw_version":"2.5.0"}
```

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2025-05-14:
pleas redownload the firmware from the release page.
when it doenst work, hit F12 on the browser and check the console log for errors and copy it here

pillepallepeng, 2025-05-15:
Not sure if this is what you need:

[image]
[image]

[image]
[image]

softwarecrash, 2025-05-15:
please right click on the page and view source code, copy it to a txt and upload it here,
btw have you tried strg+f5 for a forced reload of the page?
some script blocker activ? or active vpn?

pillepallepeng, 2025-05-15:
- CTRL+F5 does not help
- I have Pi-hole running - but as I access the Epever2MQTT locally via IP-Adress I would not expect any issue with Pi-hole.
- here is the source code:

<html>
<body>

  | <!doctypehtml><html data-bs-theme=dark lang=en xml:lang=en><meta content=width=device-width,initial-scale=1 name=viewport><link rel="shortcut icon"href=data:image/x-icon;base64,iVBORw0KGgoAAAANSUhEUgAAABAAAAAQCAYAAAAf8/9hAAAAAXNSR0IArs4c6QAAAARnQU1BAACxjwv8YQUAAAAJcEhZcwAADsQAAA7EAZUrDhsAAAHsSURBVDhPrZLLSxtRFMZHY9A2Wou0FB+oiZUWTaPEVo15qInBjamSqLFVk0rV0ke0saAmgg90o5v42NQHCmYkXehC/wbFuK//gG6yyiIki0Dg69zDEEgp46L+4HKHc8/3zTnnXg7/yf0b7O7tIZVK0XcikaBdCjKIx+MI8jzGxj+hWadHu9mC94NDlHAXZBAIrENTVw+O49LrlaYOfv8cJUnBnV9cYNjlJlGOPBfZMnnaZGrqBw6DQTH131AFLtcHEjDxQ0UBunvsMBhMWFhcoiQpyODj6Fj6rz6fH03NOmpL2/BGmEkLnZdXVGJ+YRFPnj5DKPSLxAzu5uYWkUgEXba3KHxchP39AyhVVVheXoFarYHZ0oHPX76iplYNj2cCptY29DsHRLlYQTKZxOnpGaLRKFRV1dAbjLAJhsUlZdBqX8PpfIeS0jKhCiW+e73gsmQkZpABIxaL4ff1NbXxzeNBVnYOFPmPUP3iJazWTuQ9UNCQJyYnKef4+IR0aQMGzx/Bbu/Fz+0dDA4Nw+0ewfTMLFZX1+imWIw9NEdvPzY2t0iTYeAT7r1Fb4TBaBIeUwfNRal6TiU7HH0UY2dt7WY0NulIk2HwN2y4FosV8tw8XIbDYjQTSYPw1RUC6xvU0mGQF6OZSBrcDfAHIwsaPAvZdQgAAAAASUVORK5CYII=><link crossorigin href=https://cdn.jsdelivr.net/npm/bootstrap@5.3.1/dist/css/bootstrap.min.css integrity=sha384-4bw+/aepP/YC94hEpVNVgiZdgIC5+VKNBQNGCHeKRQN+PtmoHDEXuppvnDJzQIu9 rel=stylesheet><script src=https://ajax.googleapis.com/ajax/libs/jquery/3.6.0/jquery.min.js></script><script crossorigin integrity=sha384-HwwvtgBNo3bZJJLYd8oVXjrBZt8cqVSpeBNS5n7C8IVInixGAoxmnlMuBnhbgrkm src=https://cdn.jsdelivr.net/npm/bootstrap@5.3.1/dist/js/bootstrap.bundle.min.js></script><title>EPEver2MQTT</title><body><noscript><strong>We're sorry but it doesn't work properly without JavaScript enabled. Please enable it to continue.</strong></noscript><div class="container-sm col-sm-10 col-md-6 col-lg-4"> <div class="row gx-0 mb-2"style="display: none;"id=vcc_alert><div class="alert alert-danger"style="text-align: center;"role=alert><span><b>WARNING ESP VOLTAGE TO LOW</b></span></div></div><div class="row gx-0 mb-2"><div class=col><button class="btn btn-primary"style="Display: none;"id=prevDev type=button>⊲</button></div><div class=col-9><figure class=text-center><h3 id=devicename></h3></figure></div><div class=col><button class="btn btn-primary"style="float: right; display: none;"id=nextDev type=button>⊳</button></div></div><div class="row gx-0 mb-2"><div class=col><div class=progress style=height:1.8rem;><div class="progress-bar dF"aria-valuemax=100 aria-valuemin=0 aria-valuenow=0 id=SOCbar role=progressbar style=width:0%;height:1.8rem;></div></div></div></div><div class="row gx-0 mb-2 row gx-0 mb-2 bg-light rounded bg-opacity-50 px-2"><div class=col>Device Time:</div><div class="col dF"><span id=devtime></span></div></div><div class="row gx-0 mb-2 row gx-0 mb-2 bg-light rounded bg-opacity-50 px-2"><div class=col>Solar:</div><div class="col dF"><span id=solarV></span><span id=solarA></span><span id=solarW></span></div></div><div class="row gx-0 mb-2 row gx-0 mb-2 bg-light rounded bg-opacity-50 px-2"><div class=col>Battery:</div><div class="col dF"><span id=battV></span><span id=battA></span><span id=battW></span><span id=battSOC></span></div></div><div class="row gx-0 mb-2 row gx-0 mb-2 bg-light rounded bg-opacity-50 px-2"><div class=col>Load:</div><div class="col dF"><span id=loadV></span><span id=loadA></span><span id=loadW></span></div></div><div class="row gx-0 mb-2 row gx-0 mb-2 bg-light rounded bg-opacity-50 px-2"><div class=col>Temperature:</div><div class="col dF"><span id=deviceTemp></span><span id=battTemp></span></div></div><div class="row gx-0 mb-2 row gx-0 mb-2 bg-light rounded bg-opacity-50 px-2"><div class=col>Consumed kWh:</div><div class="col dF"><span id=consD></span><span id=consM></span><span id=consY></span><span id=consT></span></div></div><div class="row gx-0 mb-2 row gx-0 mb-2 bg-light rounded bg-opacity-50 px-2"><div class=col>Generated kWh:</div><div class="col dF"><span id=genD></span><span id=genM></span><span id=genY></span><span id=genT></span></div></div><div class="row gx-0 mb-2 row gx-0 mb-2 bg-light rounded bg-opacity-50 px-2"><div class=col>CO2 Reduction:</div><div class="col dF"><span id=cored></span></div></div><div class="row gx-0 mb-2 row gx-0 mb-2 bg-light rounded bg-opacity-50 px-2"><div class=col>Input State:</div><div class="col dF"><span id=inputstate></span></div></div><div class="row gx-0 mb-2 row gx-0 mb-2 bg-light rounded bg-opacity-50 px-2"><div class=col>Charge Mode:</div><div class="col dF"><span id=chrgmode></span></div></div><div class="row gx-0 mb-2 row gx-0 mb-2 bg-light rounded bg-opacity-50 px-2"><div class=col>Load State:</div><div class="col dF"><div class="form-check form-switch"><input class=form-check-input id=loadState role=switch type=checkbox></div></div></div><div class="d-grid gap-2"><a class="btn btn-primary btn-block"href=/settings role=button>Settings</a></div><script>var onLoad=(a=>{initWebSocket();initButton()});var onMessage=(a=>{let d=0,g=`none`,l=`A  `,e=`devicename`,h=``,i=`prevDev`,m=`W  `,j=`nextDev`,p=`  `,f=`   `,t=`vcc_alert`,n=`°C  `,s=`T:`,k=`V `,q=`M:`,r=`Y:`,o=`D:`;var b=JSON.parse(a.data);if(b.hasOwnProperty(`DEVICE_QUANTITY`)){deviceQuantity=b.DEVICE_QUANTITY};if(deviceQuantity==d\|\|!b.hasOwnProperty(`EP_`+ shDevNum)){document.getElementById(e).innerHTML=`No data from Device `+ shDevNum+ f+ wifiSvg;for(var c=d;c<dataFields.length;c++){dataFields[c].style.display=g}}else{for(var c= [... cut ...] t(b).map(a=>c(a));let k=`2.5.0`.split(b).map(a=>c(a));let l=d;for(i=e;i<j.length;i++){if(j[i]===k[i]){l+=f}else if(j[i]>k[i]){l+=g}else{l+=`l`}};if(!l.match(/[l|m]/g)){console.log(`Git-Version equal, nothing to do.`)}else if(l.split(f).join(d)[e]==g){console.log(`Git-Version higher, activate notification.`);document.getElementById(`update_alert`).style.display=d}else{console.log(`Git-Version lower, nothing to do.`)}})).fail((()=>{console.log(`error can not get version`)}))}))</script> <p hidden>

softwarecrash, 2025-05-15:
found a little issue
try the bin from the zip that fit your setup

[EPEver2MQTT_d1_mini_2.5.1.zip](https://github.com/user-attachments/files/20225972/EPEver2MQTT_d1_mini_2.5.1.zip)